pascallanger/DIY-Multiprotocol-TX-Module #480: Support DumboRC X6F/X6FG

knoopx, 2020-12-14:
Hi there, just got [this receiver](http://www.dumborc.com/en/2019/07/15/x6f-receiver/), manufactured by Radiolink, thinking it would bind but it does not. My guess is that its mostly the same but just some minor differences. I just ordered a logic analyser, will report later with some captures.

Here's the official specs http://www.dumborc.com/en/2019/07/15/x6f-receiver/ and some pics of the RX.

[image]

[image]

pascallanger, 2020-12-14:
They advertise hopping on 67 channels. Radiolink is hopping on 16 channels. So not the same...

knoopx, 2020-12-14:
Yeah pretty sure it's bullshit, as you already commented here https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/386#issuecomment-662161619

pascallanger, 2020-12-15:
True, I did not remember that fact... Let's see what your dumps will give us.

knoopx, 2020-12-16:
ok, here are the captures at 16mhz. plugged both tx and rx on logic analyzer and captured at the same time (both grounds too, don't know if its necessary). rx is on spi1, tx on spi2. first i powered on the tx, then the rx. went from channel 1 up to 6 from min to max. this is my first time using a logic analyzer, please let me know if i did something wrong.

[dumbo-x6+x6fg.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5702675/dumbo-x6%2Bx6fg.zip)

knoopx, 2020-12-17:
I've been trying to figure out differences, the main packet looks very much the same:

```
7F 00 29 A4 01 00 9F EB DC AE 4C B5 7F FE FF 8F FF A3 F3 DE AE 4C B5 7F FE FF 8F FF 64 3E 00 00 74 30
```

except the second byte (req. type), which seems to be always `0x00` (instead of `subprotocol|telemetry` ). packet length and structure seems to match radiolink one.

the telemetry packet however is very confusing, it matches length but the content seems to be an exact copy of the main frame + two extra bytes (rssi/vbatt maybe?):

```
0F 00 29 A4 01 00 9F EB DC AE 4C B5 7F FE FF 8F FF A3 F3 DE AE 4C B5 7F FE FF 8F FF 64 3E 00 00 74 30 65 B1
```

knoopx, 2020-12-17:
Here's also the list of regs that get initialized:

```
ADDR: 00
AGCCTRL0: 91
AGCCTRL1: 40
AGCCTRL2: 43
BSCFG: 6C
CHANNR: 00
DEVIATN: 44
FOCCFG: 16
FREND0: 10
FREND1: 56
FREQ0: 76
FREQ1: 62
FREQ2: 5C
FSCAL0: 11
FSCAL1: 00
FSCAL2: 0A
FSCAL3: A9
FSCTRL0: 00
FSCTRL1: 06
FSTEST: 59
IOCFG0: 5C
IOCFG1: 06
IOCFG2: 5B
MCSM0: 18
MDMCFG0: F8
MDMCFG1: 23
MDMCFG2: 13
MDMCFG3: 7F
MDMCFG4: 7A
PKTCTRL0: 45
PKTCTRL1: 04
PKTLEN: 40
STROBE: SRES
SYNC0: DC
SYNC1: BA
TEST0: 0B
TEST1: 31
TEST2: 88
```

I've compared them with https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Multiprotocol/RadioLink_cc2500.ino#L121 and it mostly matches except `SYNC0/1` (`CD` /`AB`)

pascallanger, 2020-12-17:
> I've compared them with https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Multiprotocol/RadioLink_cc2500.ino#L121 and it mostly matches except `SYNC0/1` (`CD` /`AB`)

I would say that this is what's preventing multi to work... They've just swapped the nibbles so it can't communicate... Try to recompile the code with only that change.

pascallanger, 2020-12-17:
> the telemetry packet however is very confusing, it matches length but the content seems to be an exact copy of the main frame + two extra bytes (rssi/vbatt maybe?):
> 
> ```
> 0F 00 29 A4 01 00 9F EB DC AE 4C B5 7F FE FF 8F FF A3 F3 DE AE 4C B5 7F FE FF 8F FF 64 3E 00 00 74 30 65 B1
> ```

The RX just sends back what it receives, this is stupid and using bandwidth for nothing...
The 2 additional bytes are added automatically by the CC2500, they are the CRC flag (match or not), RSSI and LQI.
Nothing really usefull...

knoopx, 2020-12-17:
ok, yeah that did the trick, re-binded and working :D 

what do you think is the best approach to officially add support for this?

```patch
From e3b524eac62fd20b088d6b639b81c416425525bf Mon Sep 17 00:00:00 2001
From: Victor Martinez <knoopx@gmail.com>
Date: Thu, 17 Dec 2020 20:14:13 +0100
Subject: [PATCH] x6 support

---
 Multiprotocol/RadioLink_cc2500.ino | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/Multiprotocol/RadioLink_cc2500.ino b/Multiprotocol/RadioLink_cc2500.ino
index 7c24a6c..8b83b56 100644
--- a/Multiprotocol/RadioLink_cc2500.ino
+++ b/Multiprotocol/RadioLink_cc2500.ino
@@ -119,7 +119,7 @@ static void __attribute__((unused)) RLINK_init()
  }
 
 const PROGMEM uint8_t RLINK_init_values[] = {
-  /* 00 */ 0x5B, 0x06, 0x5C, 0x07, 0xAB, 0xCD, 0x40, 0x04,
+  /* 00 */ 0x5B, 0x06, 0x5C, 0x07, 0xBA, 0xDC, 0x40, 0x04,
   /* 08 */ 0x45, 0x00, 0x00, 0x06, 0x00, 0x5C, 0x62, 0x76,
   /* 10 */ 0x7A, 0x7F, 0x13, 0x23, 0xF8, 0x44, 0x07, 0x30,
   /* 18 */ 0x18, 0x16, 0x6C, 0x43, 0x40, 0x91, 0x87, 0x6B,
-- 
2.29.2+GitX
```

pascallanger, 2020-12-17:
I'll create a sub protocol tomorrow.

knoopx, 2020-12-17:
Awesome, thanks you! That receiver is a popular affordable choice when converting a brushed to brushless rc car.

pascallanger, 2020-12-17:
@knoopx Please try v1.3.1.89 and report. You should find a new subprotocol "DumboRC" under "RadLink".
Check also the doc: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#radiolink---74

knoopx, 2020-12-17:
eeeer I have an internal multi on a jumper t12 and the subprotocol does not show up, I guess we need opentx to be updated aswell?

pascallanger, 2020-12-17:
Which version of OpenTX do you have? The module is sending the protocols and subprotocols names to the radio. So it the telemetry is working between the module and the radio you should see them...
2 options:
 - telemetry is not working: do you see the module Status (version and channel convention) under the protocol name?
 - you are running a quite old version of OpenTX which doesn't support Multi names

knoopx, 2020-12-18:
@pascallanger sorry my mistake, I flashed the wrong binary... I'm using a fresh version but running a custom fork so I thought something was missing, anyway the new subprotocol is working! :D

pascallanger, 2020-12-18:
Sounds good, closing this issue then. Enjoy.

closed 2020-12-18 by pascallanger