pascallanger/DIY-Multiprotocol-TX-Module #487: Frsky Access

SPCORRIV, 2020-12-22:
Hey Pascal

      Any chance of adding Frsky Access so we can use The new Frsky Archer series receivers ?
Many Thanks
Scott

pascallanger, 2020-12-22:
Notthing planned.

SPCORRIV, 2020-12-22:
Hey Pascal

  Thank you for your speed response.

Thanks
Scott

salarian, 2021-01-08:
I was wondering if there is technical reason for "Won't do" or this is because of licensing/legal issues?

diego0815, 2021-01-12:
As I see it frsky is doing all to prevent cloning, so far only RXs with either cc2510 or the new unknown integrated chipset got the access firmware. So no SPI tracks to monitor...

pascallanger, 2021-01-19:
It's not really to have access to the SPI bus, it's more that I have plenty of things and this is not in my list for now.

jimsynz, 2021-02-16:
If one was to take this on themselves all one would need an ACCESS tx, rx and logic analyser, right?

kilrah, 2021-02-17:
No becasue there's no place you can tap into with it on ACCESS devices.

jimsynz, 2021-03-02:
Okay. So it's matter of trying to decode the protocol over-the-air?

demonixis, 2021-03-04:
Is there a doc somewhere to try to hack protocols? IMHO the ACCESS protocol is very important because all new FrSky receivers work with that protocol.
I got an X9 Lite and some ACCESS receiver (RXSR and RX6R), I can try few things

jimsynz, 2021-03-04:
I accidentally bought a GR6 ACCESS receiver, and am willing to buy one of the new TX's in order to help with this project. I am also willing to buy test hardware for other's who are smarter at this than me.

stanislavdavid, 2021-03-13:
There is some documentation for access #239

diego0815, 2022-07-27:
currently it looks like it will be more likely the new RXs will get an aftermarket firmware to make them speak D16 (or DSMX) rather than multi getting ACCESS. just my 2ct

hyperlobic, 2022-10-21:
> currently it looks like it will be more likely the new RXs will get an aftermarket firmware to make them speak D16 (or DSMX) rather than multi getting ACCESS. just my 2ct

Just curious, where are you getting this information on possible alternative rx firmware?

BergWolfRC, 2023-03-28:
I am very interested in seeing support for receivers that only understand the ACCESS protocol. What is needed in terms of contribution to make this happen?