pascallanger/DIY-Multiprotocol-TX-Module #488: Limit range of 2.4ghz channels FrSky uses for FHSS

Consti10, 2020-12-22:
Hello,
I am wondering if it would be possible to add an Option to limit the channels frsky hops over in the 2.4ghz band.
This way you could use [OpenHD](https://github.com/OpenHD/Open.HD) on 2.4ghz with the channel untouched by the RC link for a live video FPV feed.

Something like this has already been done for Graupner HOTT [link](https://befinitiv.wordpress.com/2015/08/26/enable-2-4ghz-rc-systems-to-work-with-2-4ghz-video-systems/) but the modifications are not practical to the average user.

I am wondering how hard it would be to add an Option to the Transmitter such that the user can select a specific 2.4ghz channel FrSky does never hop on.

pascallanger, 2020-12-22:
You can do it easily on FrSky D8 ans FrSky X v1 but not on v2.1.
This is a one line change in the code so really simple to achieve.

pascallanger, 2020-12-22:
Adding an option on the radii is another story...

Consti10, 2020-12-22:
Hi,
Can you point me to the line for FrSky ?
So far I have found it for the NRF24L01:
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/2fae0a35b833bbded1ae5309e3b0a13f9e1dfd22/Multiprotocol/BUGSMINI_nrf24l01.ino#L26

Consti10, 2020-12-22:
One more question: If I change the hopping behaviour on the Radio, is this information forwarded to the Receiver during the binding step ?

pascallanger, 2020-12-22:
Yes for FrSky D8 and D16v1, no for D16v2.1.

pascallanger, 2020-12-22:
> Can you point me to the line for FrSky ?

Which one?

Consti10, 2020-12-22:
> You can do it easily on FrSky D8 ans FrSky X v1 but not on v2.1.
> This is a one line change in the code so really simple to achieve.

Well, this one if it is one line :)

pascallanger, 2020-12-22:
> Well, this one if it is one line :)

FrSkyD: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/2a78b1d6b7a1e5837c2b448962fb17996c84e49a/Multiprotocol/FrSkyD_cc2500.ino#L106

FrSky D16v1: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/2a78b1d6b7a1e5837c2b448962fb17996c84e49a/Multiprotocol/FrSkyDVX_common.ino#L138

closed 2020-12-22 by pascallanger