pascallanger/DIY-Multiprotocol-TX-Module #490: Fly E016H with PC

Ralung, 2020-12-23:
Hello, I am new in here and I would like to ask how to fly E016H wit PC.

My purpose is to fly three E016H at the same time and let them do some simple formation mission.
But first I have to let at least one E016H be controlled by my laptop.
The following is the picture to show my idea.
[image]

Now I have Arduino Uno & CC2500 & E016HV2 protocol.
But I have no idea how to fly E016H using above things.
If anyone have any idea, please help me or give me some references and hints.
I will be really appreciate to your help!!!
(Please let me know if there is anything unclear)

pascallanger, 2020-12-23:
You already contacted me a couple of times by email. This is your school project so I'm not going to do it all for you... I've already accelerated the reverse engineering of the E016H so you can get going.
As you know the code for the E016Hv2 protocol is available on this github. Port it on your Arduino and send commands using the serial port from your PC to your Arduino.

Ralung, 2020-12-24:
@pascallanger Yes sir, thank you help and advises very much.
I am not asking you for doing whole project for me and I am very sorry that I let you misunderstand.
All I need are some guide such as what is the exact job of the protocol? What should I contain in my python code?
As a student and a beginner I really need someone to share their related experience.

Thanks again for your quick help and advises!

closed 2021-01-06 by pascallanger