pascallanger/DIY-Multiprotocol-TX-Module #491: FrSky X9Lite with TBS MPM Module - No Telemetry

mikeller, 2020-12-23:
### Setup:
- FrSky X9 Lite with custom build of OpenTX 2.3.10 (7ac1ecb6) built with `MULTIMODULE=YES`;
- TBS MPM module with `multi-stm-serial-aetr-v1.3.1.92.bin` vanilla multiprotocol firmware.

### Reproduction:
- select 'MULTI' for 'External RF' in OpenTX.

### Expected Behaviour:
- 'Status' shows the module status.

### Actual Behaviour:
- 'Status' shows 'No Telemetry'

pascallanger, 2020-12-23:
I don't know this TBS module. Was it working with the original firmware it came with?
Can you do pictures of the inside?

mikeller, 2020-12-23:
@pascallanger: It is this one: https://www.team-blacksheep.com/products/prod:tbs_mpm

The pre-installed firmware reported as 1.3.1.92 as well, and it had the same problem (RC working, but 'No telemetry') - I then did some tests with custom firmware with telemetry inversion / telemetry inversion switching disabled, without success, and flashed back to vanilla.

[image]
[image]

kilrah, 2020-12-23:
So it never worked even with the original firmware? Seems hardware related then...
Did you try stock OpenTX?

mikeller, 2020-12-23:
@kilrah: Yes, it never worked with the original firmware that the MPM came with. I did try stock OpenTX, but this does not support 'MULTI' as an option for 'External RF', so that wasn't going to work (unless I am misunderstanding something).

kilrah, 2020-12-23:
OpenTX distributed by FrSky doesn't include multi, but all downloads from OpenTX Companion do. I just meant "not compiled yourself".

mikeller, 2020-12-23:
The vanilla version that I originally tested with was the 2.3.10 (7ac1ecb6) that companion23 downloads and installs (`opentx-x9lite-afhds3-faichoice-flexr9m-lua-en-2.3.10-otx.bin`) - I am assuming that this from OpenTX and not FrSky? Looks like this version is built without support for 'MULTI'.

kilrah, 2020-12-23:
Just tried that some file and multi is here...
Have you turned off internal RF?

mikeller, 2020-12-23:
@kilrah: Interesting. I flashed vanilla OpenTX again, and this time 'MULTI' is showing up - not sure what went wrong the last time round. And yes, I'd tried with 'Internal RF' disabled both times round.
But unfortunately, the end effect is the same - 'No Telemetry'. :disappointed:

kilrah, 2020-12-23:
Telemetry will definitely not work if internal is on, but should be fine when off. Can check on my radio and a multi tomorrow but it really seems likely theres something wrong with the module...

mikeller, 2020-12-23:
@kilrah: Internal was definitely off when testing. 

> it really seems likely theres something wrong with the module...

Do you mean as in 'DOA hardware'?

kilrah, 2020-12-23:
Possibly...
2 more things to check, firstly an OpenTx build without faichoice (it's for disabling telemetry after all) and that you didn't check "disable telemetry" in the external RF settings

mikeller, 2020-12-23:
@kilrah:

> an OpenTx build without faichoice

Didn't help.

> you didn't check "disable telemetry" in the external RF settings

'No Telem' was never checked for the external module.

kilrah, 2020-12-24:
OK, so with an x9 lite with `opentx-x9lite-afhds3-faichoice-flexr9m-lua-en-2.3.10-otx.bin` and a irangex lite with `multi-stm-serial-aetr-v1.3.1.92.bin` everything is fine here, as expected no telem if internal is on and working when it's off... even when FAI mode is enabled.

[image]

mikeller, 2020-12-24:
@kilrah: Not entirely sure how your last comment relates to the problem described in this issue?
Is the 'irangex lite' the same hardware as the TBS MPM (maybe @tbs-trappy can confirm), and this is a case of DOA hardware?

tbs-trappy, 2020-12-24:
the hardware is identical to RadioMaster protocol. I am not familiar with iRangeX lite hardware. certainly not ruling out DOA, if e.g. one pin would not connect then telemetry would not show up, correct? perhaps a quick test with multimeter to test the telemetry input pin passes all the way to the top board could rule out the most likely case of the wire being faulty?

mikeller, 2020-12-24:
Thanks @tbs-trappy. Is this the correct pinout for the module port on the X9 Lite and the TBS MPM module:

[image]

Can you please let me know what pins the RC signal and telemetry are meant to be on on this connector?

tbs-trappy, 2020-12-24:
yes, we do not have a connection for 6, 7 and 8 though.

kilrah, 2020-12-24:
Telemetry comes on the S.Port pin

mikeller, 2020-12-24:
Thanks @tbs-trappy and @kilrah, I will be doing some logic analysing then.

mikeller, 2020-12-24:
Ok, this is what I get on startup:

[image]

Looks like the module sends inverted data on the S.Port pin for about 1.3 seconds, and from then on only sends periodical spikes. This is how one of the spikes looks like:

[image]

On closer look, the initial 'data' on S.Port also turns out to be only spikes:

[image]

Just for fun I uncommented `#define INVERT_TELEMETRY` in `_Config.h` and rebuilt, but even with this (resulting in uninverted telemetry) the output on the S.Port pin is just spikes:

[image]

So to me it looks like (@tbs-trappy to confirm):
- the MPM module has an external inverter on the telemetry output / S.Port pin;
- this inverter seems to be garbling the telemetry output on my module - either due to a DOA module, or through a design fault.

pascallanger, 2020-12-24:
You need to also comment this line to change the telemetry polarity otherwise OpenTX will switch it back:
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/2098cdc5e04e22e84ceb8559ceb74b2dd5d83b48/Multiprotocol/_Config.h#L301
Not saying it will help but at least you will have control.

mikeller, 2020-12-25:
@pascallanger: I have already tried that. But as the first three and the last screenshots above show, even with `INVERT_TELEMETRY_TX` the line level ends up being low in vanilla Multiprotocol, with `INVERT_TELEMETRY` not defined, or high when `INVERT_TELEMETRY` is defined.

mikeller, 2020-12-25:
I did some more testing with different combinations of `TX_INV_on/off` and `RX_INV_on/off`, and tried setting the UART TX pin to OD, without success. Without getting access to schematics for the module from @tbs-trappy I don't think there is much more that can be done to debug this.

pascallanger, 2020-12-25:
Check what you have at the input of the inverter (C86) and output.

mikeller, 2020-12-25:
@pascallanger: Is this on the TBS MPM board? Mine does not have component numbers on the silk screen, do you have a picture / diagram?

pascallanger, 2020-12-25:
This is the component marking, small chip right next to the STM 32

mikeller, 2020-12-25:
@pascallanger: I suspected so. I am currently mobile and I don't think I'll be able to probe this one without shorting and breaking it - do you have an engineering sample of this module?

pascallanger, 2020-12-25:
Like I've said at the beginning of this thread i've never seen this module... And I have no TX with this connector either...

tbs-trappy, 2020-12-29:
please let me know if you need an engineering sample. I've asked RadioMaster (the designer and manufacturer) to look at this page however I have not yet received a response from them.

pascallanger, 2020-12-29:
Engineering samples would be great in any case.
But I mean you must have tested this module type before selling it on a radio. Was it working? What was the setup?

mikeller, 2020-12-29:
Happy to do some more testing with this, but I am currently traveling and have limited access to testing gear. I'll be able to do more in-depth testing after 18 Jan.
Probably a good question to ask RadioMaster is if they have specifically tested telemetry feedback (can be verified by getting the multiprotocol version to show under 'Status' in in the OpenTX module configuration), and what TX model(s) they have tested this with - one suspicion is that this could be caused by different electrical properties (pull up / down) on the SmartPort pin on the TX module.
Just to recap, what TX models is this module designed to work on:
- TBS Tango 2?
- FrSky XLite?
- FrSky X9 Lite
- ?

tbs-trappy, 2020-12-29:
@mikeller @pascallanger please send me an email I will make sure to send you a unit each. 

We have verified telemetry working. I've personally flown a few packs with a whoop, tested the "feel" and checked the telemetry screen on the Tango 2. I don't own a X9/Xlite, but I don't see any reason why it shouldn't work there either. Although the module is predominantly for TBS Tango 2 customers, it's supposed to work with other remotes sporting the same connector.

I will push RM's engineers again to see if I can get a response, in the meantime thanks for looking at this and I'll wait for your email to send out a unit to you guys.

kilrah, 2020-12-29:
The module only being tested on the Tango and there being a drive strength issue or such with the X-Lites was also my guess, AFAIK RM don't have a Lite module yet so nothing they could base themselves on that they know is OK with those.

mikeller, 2020-12-29:
@kilrah: Thanks for the insight. With 'drive strength issue' do you mean that the pull ups / pull downs on the TX aren't strong enough? Also, to make my testing a bit less potshot-y, do you know if the X9 Lite expects the input on the S.Port pin to be inverted (low at rest) or uninverted (high at rest)?

kilrah, 2020-12-31:
> Thanks for the insight. With 'drive strength issue' do you mean that the pull ups / pull downs on the TX aren't strong enough?

Possibly... Would be good to look at the signal with a scope to see how it looks like.
Polarity should be like any other frsky radio, so inverted TTL.

mikeller, 2021-01-01:
Good point @kilrah. I currently only have a cheapo knock-off Saleae logic analyser with me, so I can not look at the signal form - but it is well possible that the form is right but the levels are off. And thanks for clarifying the polarity - this reduces one variable in the guesswork.

pascallanger, 2021-01-06:
Short this diode and try again:
[image]
That should do the trick. If it works as I assume, replace the diode by a small resistor like 470Ohm and you are good to go.

mikeller, 2021-01-09:
@pascallanger: Did a quick test with the non-return diode shorted out, and with a 1k shunt over it. Unfortunately this did not change anything, which is what I'd expected, considering that the pull up on the S.Port pin also had not fixed it.

pascallanger, 2021-01-09:
1K is going to be too much. Just shorten the diode has shown and use an official firmware so we know what you are running.

mikeller, 2021-01-09:
@pascallanger: See my comments above.

sfjuocekr, 2021-01-15:
Do you get the with FlySky only or also with others?

I get the same no telemetry status when I switch to my cheapo receiver and I have it turned off:

[image]

Then when I connect it:

[image]

This receiver doesn't even have telemetry, so I don't really care :D

mikeller, 2021-01-16:
@sfjuocekr: This is about the telemetry between the multiprotocol module and the TX - this should be independent of the over the air RC protocol selection.

sfjuocekr, 2021-01-17:
Ah, so "No telemetry" reflects TX <> MULTI and "No MULTI_TELEMETRY" reflects MULTI <> RX?

Mildly confusing as the wiki page on the X10 suggests they are the same.

kilrah, 2021-01-17:
On the Module status line they both mean Multi->TX and nothing else.

sfjuocekr, 2021-01-17:
Ok, then what @mikeller said was a wrong assumption. Yes, this receiver has no telemetry but it wasn't even bound so this status did reflect a lack of telemetry from the module.

I get the MULTI_TELEMETRY status on first boot up sporadically and only noticed it with FlySky2A selected.

kilrah, 2021-01-17:
>  Yes, this receiver has no telemetry but it wasn't even bound so this status did reflect a lack of telemetry from the module.

That's exactly what he said...

sfjuocekr, 2021-01-18:
I know, but I guess the "No MULTI_TELEMETRY detected" in my example is then just a freak coincidence if it has nothing to do with with telemetry to the RX?

It goes away if I switch protocols, most of the times it works it just shows "No MULTI_TELEMETRY detected" sporadically.

kilrah, 2021-01-18:
It has nothing to do with telemetry to RX indeed and that's what we've all been saying.

ryancummings, 2021-01-22:
Just dropping in to report I'm having the same "No Telemetry" error using the TBS MPM with my X9 Lite S. The message is present regardless of module protocol or rx bind.

tbs-trappy, 2021-01-22:
we've just had a closer look into this. 

our test units are indeed working, however they are on firmware v1.3.1.59. It seems that the latest units are shipping with a more recent firmware which does not work.

pascallanger, 2021-01-22:
> we've just had a closer look into this.
> 
> our test units are indeed working, however they are on firmware v1.3.1.59. It seems that the latest units are shipping with a more recent firmware which does not work. Do you see any changes that could be the cause? The latest firmware is also not working on Tango 2, btw.

That's more than surprising and hard to believe... There has been no change from the protocol side for at least a year and it was 1.3.0.44.
Since you are able to test, try all the releases and find out at which point it breaks. That's easy to do and will help to narrow it down.

tbs-trappy, 2021-01-22:
@pascallanger I've noted above please dm me your address or send me an email (trappy & gmail) and I'll send you whatever you need. 

I'm not familiar with the details of the issues, I'm simply providing the information I've received: 
* Tango2, with version 1.3.1.59 or below, works
* Tango2, with version beyond 1.3.1.59, does not work

I'm waiting on a X-Lite and will provide the same data points. Also, this is not TBS "hardware" per se, it's RadioMaster hardware in a TBS enclosure, which makes this even more puzzling. I've tried contacting RadioMaster but I have not yet received a response.

kilrah, 2021-01-22:
On the OpenTX side we've got Tangos but they're the old version without the module port.

pascallanger, 2021-01-22:
@tbs-trappy I've sent you an email. I'll start to investigate what could be the problem with 1.3.1.59+

mikeller, 2021-01-25:
Thanks guys for debugging this.
I did a quick check with the X9 lite, but 1.3.1.59 still gives me 'No Telemetry'. I tried both `multi-stm-opentx-aetr-inv-v1.3.1.59` and `multi-stm-opentx-aetr-noinv-v1.3.1.59`, with OpenTX 2.3.11.

I'll let you guys first work out what is wrong with newer versions on TBS hardware. Once this is sorted we can look into making non-TBS hardware work with the fixed version.

ryancummings, 2021-01-25:
Likewise, 1.3.1.59 is still showing "No Telemetry" on my X9 Lite S.

pascallanger, 2021-01-31:
@tbs-trappy
As you know the issue you raised was not a "real" issue since you were using a really old version of Flash Multi to update the firmware. As pointed out in the release notes of v1.3.1.65, in order to gain 5KB of flash space we have changed the tooling which means that you must use one of the most recent version of FlashMulti (v0.6.1 currently) to flash any v1.3.1.65+ firmwares.
To make it more obvious I've added the Flash Multi version to use in each release notes from v1.3.1.65 and up.

pascallanger, 2021-01-31:
@mikeller
I do have a FrSky X9 Lite to test with a TBS module.
1. The TBS module due to the case does not fit in the X9 Lite... You have to open the module to connect it. For a proper usage you surely need to make/print some parts to get it to fit permanently on the radio. I think TBS should add a note/warning on their selling page.
2. The telemetry does not work right away due to a high load from the radio.
3. I confirm that my previous advice to short the output diode works:
  - [image]
  - [image]
  - As you can see I've used a simple plier to do the short:
  - [image]
  - [image]
4. Adding a 1K resistor in parallel of the diode makes the trick as suggested previously:
[image]

ryancummings, 2021-01-31:
Confirmed working on x9 lite s with multiprotocol firmware 1.3.1.92 after adding a 1k resistor in parallel with the output diode as above. Thanks!

[image]

mikeller, 2021-02-01:
@pascallanger:

1. I think TBS is making it plenty clear that an adapter is needed:

[image]

2. Can you define what you mean by 'high load from the radio'?
3. it did not work for me when I tried it last time - but at that time I was using newer 1.3.1.92 multiprotocol firmware that still seems to not be working with this hardware (I used `flash-multi-0.6.0`, which was the latest version available when I downloaded the multiprotocol firmware). I will test again, but as said before, if this works with older firmware then the solution of adding an additional pull up that - unlike shorting out the diode - does not make it impossible to use the bidirectional capabilities of the S.Port pin should work as well, and will be preferrable as a permanent solution (could probably be included on the adapter board that comes with the X9 case for the TBS MPM).

[4 more replies]

closed 2021-02-01 by pascallanger