pascallanger/DIY-Multiprotocol-TX-Module #495: Support Easycopter V4 COLIBRI

valfa14, 2020-12-28:
Hey, I've tried to connect Easycopter V4 COLIBRI to J6pro but without success.
Receiver is NE260A RX-06, TXs TX16s/TX18s

[image]

Any hints?

pascallanger, 2020-12-28:
No idea... If you have tried all protocols and sub-protocols then the next step is to open the radio to see what's inside by taking good quality and close up pictures of where the antenna is connected.

stanekTM, 2020-12-28:
I have an identical transmitter, it should contain CYRF6936 and DSM protocols.

valfa14, 2020-12-28:
[image]

valfa14, 2020-12-28:
[image]

stanekTM, 2020-12-28:
Oh, the same packaging and different electronics.

pascallanger, 2021-01-06:
If you are willing to reverse this protocol, you can connect a SPI analyzer on the RF board and do some dumps.

pascallanger, 2021-01-19:
@valfa14 Should I close this issue?

closed 2021-02-08 by pascallanger

valfa14, 2021-02-08:
@pascallanger Sorry, I didn't have time to provide you all information that you need. I will get back to this problem right after I will test all available sub-protocols.