pascallanger/DIY-Multiprotocol-TX-Module #496: Multiple receivers have the same bind information

sgmservice, 2020-12-30:
I have two airplanes; both have Orange R820x V2 Receivers with Lemon satellite receivers. The planes are set up as separate models, Schtick v3 and Schtick v4, in the transmitter and were bound individually. If I turn on both airplanes the transmitter will bind to and control both airplanes from either model configuration. I have re-bound several times and get the same results 

The transmitter is a Jumper T16, Open TX V 2.3.10, internal multiprotocol module firmware version 1.3.1.69. I have nine models set up in the transmitter. The receivers are a mix DSMX (Orange and Lemon) and Frsky D16 (Frsky and Jumper). From what I have seen only these two models are sharing bind codes.

benlye, 2020-12-30:
Give each model a unique receiver number and rebind them both. 

https://www.multi-module.org/using-the-module/receiver-match

sgmservice, 2020-12-30:
Ben,

Thanks for the quick reply. My bad! Should have check that first. I was rearranging “Groups” on the radio and inadvertently duplicated the Model Number.

Thanks for all of the hard work. I absolutely love the combination of Open TX and  Multimodule.

 

Again, Thanks

Gene Maurice

 

https://www.multi-module.org/using-the-module/receiver-match

closed 2020-12-30 by benlye