pascallanger/DIY-Multiprotocol-TX-Module #500: FS2A RX "FAIL"

Franciso81, 2021-01-11:
Buenas noches, quería agradecer el trabajo que realizan para que sea posible esta gran herramienta que han desarrollado.
Estoy intentando enlazar una emisora Flysky i6x al modulo multiprotocolo 4en1 de una TX16S para hacer "Wireless trainer" sin éxito. incluso con otro modulo y seleccionando el protocolo FS2A TX. 
El único sistema que he conseguido enlazar es el FRSKY X. 

Un saludo y gracias de nuevo

pascallanger, 2021-01-12:
Can you post in English please?
I'll check the RX code and come back to you.

pascallanger, 2021-01-12:
Fixed in v1.3.2.2. You can test with the version available here: https://downloads.multi-module.org/latest-test/

Franciso81, 2021-01-12:
Good evening, excuse my English but I have to use the google translator.

I have installed the firmware you sent me (v1.3.2.2) in the multiprotocol
and it seems to link, but it does not finish. I don't know if it may be the
telemetry or the
channel information that is failing. I have recorded a video so you can see
what happens and you can get a better idea.
 20210112_194426.mp4
<https://drive.google.com/file/d/1L4eySzzmZzJ828TtTuh4oAVDM_maB7XB/view?usp=drive_web>

Thanks for your work and the rest of the team and here you have me for what
you need.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Libre
de virus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#m_8718707613971360483_DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Libre
de virus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

pascallanger, 2021-01-13:
The TX and RX must hear each others. Try to bind at different distances. Like really close nearly the antennas touching to a few meters.

Franciso81, 2021-01-13:
ok, this afternoon I will do new tests and I will answer you with the
results. thanks

Franciso81, 2021-01-13:
Good afternoon, I have tried to make the link at various distances and
turning off the LNA without success.
There is communication because the station with the module (RX) stops
linking but it seems that it does not send a response or this is wrong,
since
the original Flysky transmitter does not end. with an original receiver if
it binds perfectly.

Thank you

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Libre
de virus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Franciso81, 2021-01-13:
With which teams do you link? and does the wireless trainer mode work for you?
the Flysky station is bought at Christmas this year but I don't think that's the problem.

pascallanger, 2021-01-14:
I don't have an original Flysky radio so I can only test between 2 multi modules and it works fine here. I'll check what I can do.

pascallanger, 2021-01-14:
I understand that the original radio does not exit bind. But have you tested if the trainer actually works in the end? From a multi perspective, the fact that it exits means that everything has been negotiated and it has received the confirmation from the TX...

Franciso81, 2021-01-14:
Good morning, I have tried in several ways and there is no way for the coach to function with this protocol. with Frsky if it works perfectly. Apart from having tried the original Flysky radio, I also have two multiprotocol modules and if I link between them, the two end the link but when activating the trainer, it sends channels in a strange way and without obeying the sticks and with RSSI-0 measurement. On both I have put the firmware 1.3.2.2 that you sent me.
Thank you.

pascallanger, 2021-01-14:
Oh, ok I've tried only the bind between the 2 modules... Let me try the full stuff.

sfjuocekr, 2021-01-15:
Binding works fine with 1.3.2.2, but I'm not getting any channel data. All channels are set to -125 if I check the trainer tab.

When I switch it to FrSky RX it works as intended.

Now that I think of it, Im still running 1.3.1.92 on the slave and 1.3.2.2 on the master. I'll upgrade the slave to see if it makes a difference.

pascallanger, 2021-01-16:
I haven't had time to check yet...
Upgrading the slave shouldn't make a difference.
I'll try to have a look today.

pascallanger, 2021-01-16:
This time it should be all fixed in v1.3.2.5 available as a test build here: https://downloads.multi-module.org/latest-test/

Franciso81, 2021-01-16:
Sorry but the link you gave me is version 1.3.2.4, in case it was a mistake. I have tried this one and it does exactly the same thing,
TX Flysky-multi does not link and multi-multi links but without channel control and RSSI 0.

pascallanger, 2021-01-16:
I don't know why the page link has not updated itself automatically...
Here is the direct link: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/suites/1842296535/artifacts/35655530
version should be 1.3.2.5

Franciso81, 2021-01-16:
Ok, this is going in the right direction. Now the between multi-multi works fine, but not Tx Flysky-multi. there is some step that the original receiver sends that the multiprotocol does not. in both cases, the RSSI is still 0. thanks

Franciso81, 2021-01-16:
I think you don't have an original Flysky station. Is there any way that I can send you data that you need so that you can see what is happening? I tried to install the xn297dump-usbdebug firmware and I see that it does channel scanning but it does not capture anything. I don't know if this is the correct way to do it.

pascallanger, 2021-01-16:
XN297dump won't do anything...
I would need a SPI dump of your original TX binding a RX and binding with Multi to see the difference.

pascallanger, 2021-01-16:
@goebish is this something you can help with?
Basically it works between 2 multi modules but not with an original radio or at least not that one. Do you have a TX to test?

pascallanger, 2021-01-16:
I'm wondering if this is only with this original TX or all of them... Can someone else do a test?

goebish, 2021-01-16:
I've an original AFHDS-2A TX, I'll try tomorrow

Franciso81, 2021-01-16:
Yes, I have an original Tx Flysky I6X and IA6B receiver that I have borrowed for testing purposes. When I link them, the Tx shows "Bind OK" on the screen and exits to normal mode, while when I link it with the multiprotocol, the Tx does nothing but the multiprotocol does exit bind mode

pascallanger, 2021-01-16:
What I can try tomorrow as well would be to bind multi in TX mode with an original RX to see what the RX replies and how it differs from what we send back in RX mode.

pascallanger, 2021-01-16:
> I've an original AFHDS-2A TX, I'll try tomorrow

You need the latest multi version from master, previous were not working at all...

benlye, 2021-01-16:
> I don't know why the test page link has not updated itself automatically...

Seems to be a Github back end issue. The AFHDS 2A RX build doesn't show up in a query for builds with a status of 'success', even though its completion status actually was 'success'. 

https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/actions?query=event%3Apush+is%3Asuccess+branch%3Amaster

Very weird.

pascallanger, 2021-01-16:
> Seems to be a Github back end issue. The AFHDS 2A RX build doesn't show up in a query for builds with a status of 'success', even though its completion status actually was 'success'. [...]

Thanks for looking into it. Anyway it's great that you can just link the build files directly from the job :-)

sfjuocekr, 2021-01-17:
1.3.2.5 works and now shows values on the trainer tab after binding.

pascallanger, 2021-01-17:
> 1.3.2.5 works and now shows values on the trainer tab after binding.

Which TX are you using to bind?

sfjuocekr, 2021-01-17:
Nope, I was binding between two multi modules. Two TX18S radios.

pascallanger, 2021-01-17:
> Nope, I was binding between two multi modules. Two TX18S radios.

ok so we know that this is working.
Now we need to figure out what's not working with a genuine radio.

pascallanger, 2021-01-18:
Looking at the betaflight code which has support for afhds2a spi receiver, I see that the bind is handled differently. I think we lack the final response to the TX to complete the bind. Not having an original TX is not ideal to troubleshoot this, I can try to change the code but... I'll ask around to see if I can borrow one.

goebish, 2021-01-18:
Sorry I didn't have time to search and charge mine :(

Franciso81, 2021-01-18:
OK, that would be ideal. However, any modification you make I can test and see if it works.

goebish, 2021-01-18:
You're probably right about the final RX response, if there's one I probably never implemented it in the TX part since it's not required for the protocol to work.

pascallanger, 2021-01-19:
@Franciso81 please test the version 1.3.2.8 : https://downloads.multi-module.org/latest-test/
Make sure to test a few times since I've seen the bind failing may be once every 10/15 trials.

Franciso81, 2021-01-19:
Ok, I have tested this firmware and you are right, sometimes it makes the TX go out of bind mode and other times it does not (one out of every 4 times more or less).
the trainer mode works but not always that it links, sometimes it leaves the bind mode and it does not work and other times it does. Now we just need to refine a little that we can say that it works.

Franciso81, 2021-01-31:
Good morning, yesterday I was able to test the wireless student-teacher function and everything went well, surprisingly I was able to link the original station to the first one and it worked well but only with the LNA activated (unmarked). If I separated a little from the student, he lost signal with the multiprotocol. but keeping together (2 meters) everything was fine. When detecting this, I linked again and again I was able to do it the first time, but it was all the same. 
In conclusion, this function can be used but with limitations.
A greeting

pascallanger, 2021-01-31:
That's expected since the 2 antennas are most likely really close to each others. Try to move them around and put as much distance between the antennas as you can.

closed 2021-01-31 by pascallanger