pascallanger/DIY-Multiprotocol-TX-Module #502: FrskyX2 with XM+

sfjuocekr, 2021-01-15:
I have two receivers, one R-XSR which works flawless and one XM+ that has the following "problem":

When I put it in bind mode and receiver is set to anything but zero it won't bind at all, if I put it to zero it binds no problem.

So I flashed back the XM+ to 1.x firmware and tried again with FrSky instead of FrSkyX2 and the result was the same. If I try to bind the R-XSR it always works.

I'm not sure if this is an issue specific to the XM+, but I remember having the same problem with my previous transmitter (IR8M).

r41065, 2021-01-19:
looks that there is a more than one problem with XM(+) v2 Firmware:
They are not correct working even with original Transmitters.
( v2.1.0A from Github works but has some Failsafe Issues;
 v2.1.2 from FrSky Website works only if Distance betwen TX and RX is bigger than 3m )
v1 is working fine.
Ralf

pascallanger, 2021-01-19:
Issue probably related to the RX

closed 2021-01-19 by pascallanger