pascallanger/DIY-Multiprotocol-TX-Module #504: Turnigy 9X Telemetry Mod: Ambiguous Pins

SomeInterestingUserName, 2021-01-18:
Concerning the section on the Transmitters page for performing the [9X telemetry mod](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Transmitters.md#Telemetry_Mod):
It appears that the two linked guides for the FrSky telemetry mod list different pins (Transmitter MCU Pin 2 vs. Pin 3) to connect to the Multiprotocol module's Pin 5 telemetry output, which might lead to some confusion. In my experience, I was only able to get telemetry (and the module itself) working when I connected the Multiprotocol module's Pin 5 to the transmitter MCU Pin 2. 

It might also be worth pointing out that if the telemetry output was wired incorrectly, the Multiprotocol module will not continue past the bootloader (some rapid flashes from the red LED, and then no light). This could help with troubleshooting a miswired transmitter.

For reference, this behavior occurs on Multiprotocol firmware version v1.3.1.92, hardware Jumper JP4IN1-SE, although I suspect it may happen on other arrangements as well. I was using a Turnigy 9X radio with er9x.

To clarify, this doesn't appear to be an issue with the Multiprotocol firmware, just a suggestion to add some troubleshooting info on the telemetry mod guide.

pascallanger, 2021-01-19:
I've updated the page. let me know if this is better now.

closed 2021-01-19 by pascallanger

SomeInterestingUserName, 2021-01-19:
Thanks for adding the troubleshooting tip! 

One more thing--the Oscar Liang page linked has nice detailed instructions, however it seems to have the wrong pin for the Pin 5 telemetry (TX Pin 3 instead of 2). A comment on that site mentions this, although I'm not sure if it's worth including a note on here as well.