pascallanger/DIY-Multiprotocol-TX-Module #505: FrSky SPI - Telemetry lost when moving sticks

AndreaCCIE, 2021-01-19:
Hi all, I have a weird issue that I have no idea if it's up to the SPI (that has worked correctly for a while), OpenTX or the multiprotocol module on my TX16S.
It's related to the only 65mm whoop I do fly, and it's been on a shelf for a while so I can't say if it's related to a multi protocol or OpenTx specific update.

Basically, if I move the pitch OR roll over the 1900 point when the throttle is over 1900, the connectivity to the SPI drops:

https://www.youtube.com/watch?v=Jn5d_tS4yfw

Any troubleshooting I can do to understand if it's multi protocol related?

benlye, 2021-01-19:
Does connecting the quad to BetaFlight Configurator reveal anything (assuming it's running BetaFlight)?

I think I saw something like this before on a BetaFlight whoop and it was to do with the signal exceeding a threshold in BF when I pushed a stick to the limits.  I think that only happened on one input though, not a combination like you have.

AndreaCCIE, 2021-01-19:
Everything looks good, I did test both BF and Emu with the same results. If I fly straight, there’s no problem :-)

benlye, 2021-01-19:
Have you tested it in flight or just on the bench? Does putting more distance between the TX and RX help?

AndreaCCIE, 2021-01-19:
> Have you tested it in flight or just on the bench? Does putting more distance between the TX and RX help?

I noticed the problem in flight, but I didn’t understand it was related to the sticks position.
My living room is not huge, but I had a failsafe in a meter or two.

I have been able to walk around the house holding the radio without any issue, but I wasn’t moving the sticks.

AndreaCCIE, 2021-01-20:
A second video, hoping it can be helpful:

https://www.youtube.com/watch?v=iX6-2IBSHu8

benlye, 2021-01-20:
I'm wondering if it's a BF failsafe kicking in.

It's hard to see exactly, but it looks like BF is seeing roll > 2000.  Do you have any rate settings that would be increasing the roll output when throttle is high?

I'd try adjusting output weights on the radio so that max/min pitch/roll are at 1000/2000.

Otherwise, I'd suggest setting up a new model from scratch on the radio to see if that has the same problem.

Bottom line, at this point I don't think this is a MULTI-Module issue.

closed 2021-02-01 by pascallanger