pascallanger/DIY-Multiprotocol-TX-Module #507: full futaba support

medarin, 2021-01-26:
is it possible to add futaba protocols  like the FASSTest , FASST, and T-FHSS?

kilrah, 2021-01-26:
No, none of the chips in the MPM can do those.

diego0815, 2021-01-31:
@kilrah that's for sure  true for fasst(est), but I think t-fhss is a cc2500 protocol and thus feasible if someone can produce the SPI traces??

medarin, 2021-01-31:
the T-FHSS is the main one i would want just to have the telemetry

pascallanger, 2021-02-01:
Fine to look at T-FHSS but I need a TX and RX to work on it. So if someone is willing to do this then PM me..

diego0815, 2021-02-01:
Unfortunately I couldn't find any t-fhss gear on second hand market, especially not in any sensible budget. Else I would have got a set to play with the logic analyser, or worst case ship it to F.

Doums33, 2022-07-26:
Hi, i have a t-fhss receiver that i can send by postal way to Pascal if it can help. Let me know if it could help.

raphaelcoeffic, 2026-10-29:
The RF chip is almost certainly an ML2730, a 2.4 GHz FSK transceiver now sold under Qorvo. It comes from the same “ML27xx” lineage Futaba has used before. Forum users who looked at FCC filings report that FASST uses Micro Linear’s ML2724, while S-FHSS uses TI’s CC2500. The ML2730 would be the natural successor for a newer receiver, likely a FASSTest one given the RX62T and the 2015/2016 date codes.

raphaelcoeffic, 2026-10-29:
@pascallanger FAST and FASTest could potentially be emulated based on TI's RF core (CC26xx / CC27xx).

raphaelcoeffic, 2026-11-03:
I have a working link between a [LaunchpadXL CC2652R1](https://www.ti.com/tool/LAUNCHXL-CC26X2R1) and a Futaba R6008HS FASST receiver :-)

raphaelcoeffic, 2026-11-03:
If you guys wanna help with FASST / FASSTest, you can grab a [Sonoff ZBDongle-P](https://sonoff.tech/en-de/products/sonoff-zigbee-3-0-usb-dongle-plus-zbdongle-p) (**"-P"**, not "-E"!) which also contains TI's CC2652. I wrote a firmware for it that should allow us to trace packets over the air. It can also be used then as a MPM for bench tests that can be driven from Python scripts.

*Stay tuned for more...*

diego0815, 2026-11-03:
Very interesting...

I don't own any FASST gear, but I am very much interested in finding out what all can be emulated by the Ti TRX, I heard rumours DSMX could also be emulated by the 2652, so maybe the next gen of the 4in1 needs only one single chipset...

raphaelcoeffic, 2026-11-03:
>  I don't own any FASST gear, but I am very much interested in finding out what all can be emulated by the Ti TRX, I heard rumours DSMX could also be emulated by the 2652, so maybe the next gen of the 4in1 needs only one single chipset...

DSM/DSMX already works here on my dev boards (CC2652 and CC2755). In fact, I can already emulate right now all the 4 chips present on a 4-in-1.

diego0815, 2026-11-03:
That's really cool thanks for the heads-up and the good work

diego0815, 2026-11-03:
Checked out your mpm2 repo, very cool, especially thinking about using the zigbee usb sticks for whatever you can imagine. Potentially even on an edgeTX as an external module.

raphaelcoeffic, 2026-11-04:
> Checked out your mpmv2 repo, very cool, especially thinking about using the zigbee usb sticks for whatever you can imagine. Potentially even on an edgeTX as an external module.

Once things are working well enough (tested?), a new MPM module will be manufactured and sold (not by us, but we're talking to a specific manufacturer who's interested). AFAIK, @pascallanger will probably publish schematics as well (open hardware; reference design).

raphaelcoeffic, 2026-11-04:
As promised, more on the [Sonoff ZBDongle-P](https://sonoff.tech/en-de/products/sonoff-zigbee-3-0-usb-dongle-plus-zbdongle-p) (**"-P"**, not "-E"!): it can now record FASST packets, set up from a web page, with nothing to install:

**https://fasst-sniffer.edgetx.workers.dev**

The page loads the sniffer firmware onto the dongle (about 20 seconds, through the CC2652P's ROM boot loader, like Sonoff's own flasher), then guides you through a few short recordings (sticks centred, each channel end to end, switches, off and on: about 3 minutes in total) and saves them into one text file. It needs Chrome or Edge on a computer. The page can save the dongle's current firmware first, and Sonoff's web flasher puts a Zigbee firmware back afterwards.

FASST 10ch (TM-10 / TM-14) is decoded. The sniffer catches it on all 36 hops and follows the hopping. **What I'm missing is FASST MULT and 7CH:** if you have a radio that sends either mode (8FG, 10C, 12FG, 14SG, TM-8...), a recording posted here would help a lot. Please say which radio and which mode in the file (the page asks).

Only tested in Chrome on macOS so far: if you're on Windows or Linux, please tell me whether loading the sniffer worked.

FASSTest is next: I'll have a T14SG and a FASSTest receiver to look at in a few days. If you have T-FHSS gear, say so here: I can't do anything for it yet, but that's the next one after.

The page and the firmware are in the [mpmv2 repo](https://github.com/raphaelcoeffic/mpmv2) (`tools/sniffer_web/`, `src/boards/zbdongle_p/`).

raphaelcoeffic, 2026-11-04:
Small T-FHSS update: Futaba's R334SBS V5.0 update file (receiver, T-FHSS / T-FHSS SR / S-FHSS) uses the same XOR scheme as the R7206SB one, and this time the radio code is inside.

- **RF chip:** CC2500 (26 MHz).
- **T-FHSS PHY:** MSK at 384 kbps, sync `D3 91` (same as S-FHSS), 4 preamble bytes, 14-byte packets, no hardware CRC.
- **Channels:** 31, at 2407.5 + 2·(n−1) MHz (2407.5..2467.5).
- **Hop sequence:** row 0 = `[31] + [3^i mod 31, i = 0..29]`, row r = row 0 + r (mod 31). A Welch-Costas sequence.
- **Packet:** 12 bytes XORed with a key made of the TX ID and the current hop channel (like FASSTest), then CRC-16/MCRF4XX over the plain bytes, low byte first.
- **Servo frame:** 4 × 12-bit channels plus one extra channel rotating over 4 slots.
- **Telemetry:** the receiver answers with 14-byte packets, same CRC, permuted key.
- **Special mode:** the TX can switch the receiver into a mode with a 16-channel hop list (channels 1..25 only, top ~12 MHz unused) and unscrambled telemetry. We don't know what it's for yet (France/area setting? SR?).

Still open: frame timing, how the hop row is chosen, and that special mode. A TX capture would settle most of it. With the PHY known, the ZBDongle-P can now listen to T-FHSS directly. @tegots123 (#1146), your 4PM+ with T-FHSS and T-FHSS SR receivers would be ideal for this; I'll post a T-FHSS version of the sniffer page when it's ready.

raphaelcoeffic, 2026-11-04:
Follow-up on T-FHSS: the PHY found in the R334SBS firmware is now confirmed on the bench. A 4-in-1's CC2500, set up with the receiver's exact registers, and our CC2652 (ZBDongle-P) exchange T-FHSS-format packets both ways with byte-exact payloads (384 kbps MSK, sync `D3 91`, 14 bytes, 31 channels at 2407.5 + 2·(n−1) MHz). So the 4-in-1's CC2500 can do T-FHSS too; what's missing for a transmitter is the frame timing, how the hop row is picked, and the bind details.

The sniffer page now has a T-FHSS mode: **https://fasst-sniffer.edgetx.workers.dev** (pick "T-FHSS" in step 3; if you used the page before, it will ask you to load the sniffer again). It records on one channel at a time, which is enough to get the timing and the hop order. If your radio has an Area setting (General / France), please say which one, and if you can, record both: there is a special 16-channel mode in the receiver and we don't know yet what turns it on.

@tegots123: your 4PM+ with T-FHSS and T-FHSS SR receivers would be perfect for this. One run per mode (T-FHSS, then SR), including the optional "link a receiver" step, would already tell us a lot.

tegots123, 2026-11-04:
does it need to be a CC2652P sniffer? I dont think I have one laying around... but I for certain have a generic one zigbee generic one

raphaelcoeffic, 2026-11-04:
@tegots123 CC2531 is a totally different chip, that won’t work. The Sonoff dongle is rather cheap though and useful in other contexts as well, for instance for smart home (Zigbee, Thread, etc).

raphaelcoeffic, 2026-11-04:
@tegots123 Thanks! Good news: the sniffer records the **transmitter**, so your 4PM+ and the dongle are all that's needed. No need to buy a T-FHSS receiver.

From the R334SBS firmware, T-FHSS SR uses the same radio settings as T-FHSS (the receiver has no other radio setup outside its factory test mode), so the page's T-FHSS mode should hear SR as well. When the dongle arrives:
- set the 4PM+ to SR and do the guided recordings, with your SR receiver in the "link" step;
- if the 4PM+ lets you switch to plain T-FHSS, do a second run in that mode too, even without a receiver (just skip the link step). Comparing the two is exactly what we need.

Correction to my first comment: the R334SBS is a T-FHSS / T-FHSS SR receiver, not S-FHSS.

tegots123, 2026-11-04:
if R334SBS works with both, cant i just try to sync it using tfhss instead of the SR? 

for FASST do you need to have a 7/10ch radio ? i'm certain I have a 3pm available but not if i still have a 7

raphaelcoeffic, 2026-11-05:
@tegots123 The sniffer page (**https://fasst-sniffer.edgetx.workers.dev**) now has two more choices under "System", both quick and both useful with the 4PM+:

- **S-FHSS (reference):** a protocol we already know well (MPM supports it). Please do this run first: if the page hears S-FHSS fine, the dongle and the page work on your side, and anything odd in the T-FHSS runs comes from T-FHSS itself.
- **Mini-Z (4PM Plus):** listens for the bind (link) packets of Kyosho FHSS (MPM's Kyosho protocol, FHSS sub-protocol, as used by the KT-531P). If the 4PM+'s built-in Mini-Z Evo mode is that protocol, the page will count them and MPM already covers it; if it records nothing, it's something else. No car needed: set the radio to Mini-Z Evo and start its link function during the first two steps.

So the full list when the dongle arrives: S-FHSS, then T-FHSS SR (with the SR receiver in the link step), then plain T-FHSS if the radio allows it, then Mini-Z. Each system is saved as its own file (the page asks before switching), so please save after each one.

Everything was checked here with a 4-in-1 playing the transmitter, so the only unknown left is your radio.

raphaelcoeffic, 2026-11-05:
> if R334SBS works with both, cant i just try to sync it using tfhss instead of the SR?

Just try the page with the 4PM+ and the receivers you have as described above.

> for FASST do you need to have a 7/10ch radio ? i'm certain I have a 3pm available but not if i still have a 7

For FASST, I'll probably have everything I need in a couple days: someone is lending me a 14SG and another friend has a couple receivers I can test with as well.

raphaelcoeffic, 2026-11-08:
> if R334SBS works with both, cant i just try to sync it using tfhss instead of the SR?

I checked the R334SBS manual and it seems to support both modes:
[image]

Also, the 4PM+ supports these 2 modes so you can probably just make different traces with the different modes.

raphaelcoeffic, 2026-11-08:
@tegots123 An update on T-FHSS before your dongle arrives.

The R334SBS firmware gave us more than expected:
- **Every packet carries its hop sequence number unencrypted**, so the sniffer can now **follow the transmitter's hops** across all 31 channels, like it does for FASST. Before, it only listened on one channel and caught about 1 packet in 31.
- **T-FHSS vs SR is chosen when the receiver is linked**: the transmitter sends a mode bit in the bind packet and the receiver stores it. In one mode the receiver uses a shorter frame and sends no telemetry back, which fits SR.

What we still need from the air is the real frame timing, and which mode is which. The page is updated at **https://fasst-sniffer.edgetx.workers.dev** (version 4). If your dongle already has the older sniffer on it, the page offers to update it.

When the dongle arrives:
1. **S-FHSS (reference)** first, as before: it checks that the dongle and the page work on your side.
2. **T-FHSS**, with the 4PM+ in **T-FHSS** mode: the guided steps, then the **"Link a receiver"** step with your R334SBS. That step listens on the channel where linking happens, so link the receiver during it.
3. The same again with the 4PM+ in **SR** mode, linking the R334SBS in SR.

Set the "Mode it is set to" field on the page for each run, and save one file per mode. Two runs with the same receiver should show exactly how the two modes differ.

If anyone has **T-FHSS Air** gear (e.g. a T6K, T10J or T16IZ with an R3008SB): Futaba says it isn't compatible with Surface T-FHSS, and we know nothing about it yet. A recording in the page's T-FHSS mode would show whether it's even close.

OzDruiD, 2026-11-15:
Hows this going my  - my "P" dongle is stuck and or potetnially rejected for some reason  in China customs.  I can order another or wait it out, but probabily faster to  order another from a different seller or pay to $$ via local amazon if we havent had any progress on the FHSS data?

raphaelcoeffic, 2026-11-15:
> Hows this going my - my "P" dongle is stuck and or potetnially rejected for some reason in China customs. I can order another or wait it out, but probabily faster to order another from a different seller or pay to $$ via local amazon if we havent had any progress on the FHSS data?

Nothing on **T-FHSS** yet. S-FHSS was already working, so that's not an issue.

OzDruiD, 2026-11-15:
OK what about all the others - are we fully covered for everything except T-FHSS?   eg FASSTest, FASST,  any others?
I wont have access to Surface Futaba- but possible some of the AIR ones that support T-FHSS.

raphaelcoeffic, 2026-11-15:
> OK what about all the others - are we fully covered for everything except T-FHSS? eg FASSTest, FASST, any others? I wont have access to Surface Futaba- but possible some of the AIR ones that support T-FHSS.

FASST(est) is covered, yes. I haven't seen the 24ch/36ch modes yet, but the rest just works (tested with a couple receivers).
T-FHSS still has a few unknowns and fully untested (Air and Surface; no hardware). The web sniffer should be able to follow the frequency hopping and record what I need though. Once I have the missing pieces, the same dongle can be used to test with a receiver (dongle in TX mode).

tegots123, 2026-11-15:
Just as an update, my dongle is already past local customs and will arrive tomorrow or Tuesday at max.

tegots123, 2026-11-16:
[tfhss SR-radio-2026-11-16.txt](https://github.com/user-attachments/files/34940734/tfhss.SR-radio-2026-11-16.txt)
[tfhss normal-4pm-plus-2026-11-16.txt](https://github.com/user-attachments/files/34940735/tfhss.normal-4pm-plus-2026-11-16.txt)
[sfhss-4pm-SFHSS-2026-11-16.txt](https://github.com/user-attachments/files/34940733/sfhss-4pm-SFHSS-2026-11-16.txt)

here are the files, please let me know if it works! and also i just saved the tfhss normal speed, let me know if you also need the high speed mode