pascallanger/DIY-Multiprotocol-TX-Module #508: Can't flash JP4IN1-S3, Firmware Update err

c0debreaker, 2021-01-29:
I have a Jumper T16 first model which comes with JP4IN1-SE. I upgraded from JumperTX to OpenTX last night.

I downloaded multi-stm-serial-reta-v1.3.2.12.bin and placed it in FIRMWARE folder on the sdcard. I tried flash internal and external but both failed with "Firmware update err". Then on next line, "NoSync".

I am not really sure why the bin file didn't work. I googled and most bin files I saw in forum has "opentxt" text in the filename of the bin file. However, if you look at the bin file I downloaded, it doesn't have "opentx" in the filename. I am not sure if it's critical. I just noticed it forums.

kilrah, 2021-01-29:
Most likely the firmware on your module is too old to support updating from the radio, update it and the bootloader through USB with https://github.com/benlye/flash-multi/

c0debreaker, 2021-01-29:
Ah, awesome! Thank you so much!

closed 2021-01-29 by pascallanger

c0debreaker, 2021-01-29:
I was able to flash the latest multi protocol. However, I am not sure why it's still not working in my OpenTX. This module was fine previously in JumperTX. I only upgraded to OpenTX last night. Here's a picture of what I am seeing - https://i.imgur.com/z5Vj2z0.jpg
The error is "no multi telemetry detected". I also can't bind my quadcopter.

c0debreaker, 2021-01-29:
Here's the output but I shortened it. Any ideas?
```
└> ./flash-multi -f ~/Downloads/multi-stm-serial-reta-v1.3.2.12.bin -p  /dev/cu.usbserial-0001
flash-multi 0.6.0

This program is Free Software and has NO WARRANTY.
https://github.com/benlye/flash-multi/

2021-01-29 01:46:16.484 system_profiler[73941:7576573] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2021-01-29 01:46:16.484 system_profiler[73941:7576573] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2021-01-29 01:46:16.484 system_profiler[73941:7576573] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2021-01-29 01:46:16.484 system_profiler[73941:7576573] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2021-01-29 01:46:16.488 system_profiler[73941:7576573] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2021-01-29 01:46:16.488 system_profiler[73941:7576573] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2021-01-29 01:46:16.731 system_profiler[73949:7576600] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2021-01-29 01:46:16.731 system_profiler[73949:7576600] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2021-01-29 01:46:16.731 system_profiler[73949:7576600] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2021-01-29 01:46:16.731 system_profiler[73949:7576600] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2021-01-29 01:46:16.736 system_profiler[73949:7576600] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2021-01-29 01:46:16.736 system_profiler[73949:7576600] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
Firmware File Name:        multi-stm-serial-reta-v1.3.2.12.bin
Multi Firmware Version:    1.3.2.12 (STM32F1)
Firmware Target:           STM32F103CB (128KB)
Expected Channel Order:    RETA
Multi Telemetry Type:      OpenTX or erSkyTx (MULTI_TELEMETRY)
Invert Telemetry Enabled:  True
Flash From Radio Enabled:  True
Bootloader Enabled:        True
USB Serial Support:        False
Serial Debug Enabled:      False
.
.
.
.
.
[3/3] Writing firmware...
./tools/macos/stm32flash -v -s 8 -e 0 -g 0x8002000 -b 115200 -w "/Users/rcamara/Downloads/multi-stm-serial-reta-v1.3.2.12.bin" "/dev/cu.usbserial-0001"
stm32flash Arduino_STM32_0.9

http://github.com/rogerclarkmelbourne/arduino_stm32

Using Parser : Raw BINARY
Interface serial_posix: 115200 8E1
Version      : 0x22
Option 1     : 0x00
Option 2     : 0x00
Device ID    : 0x0410 (Medium-density)
- RAM        : 20KiB  (512b reserved by bootloader)
- Flash      : 128KiB (sector size: 4x1024)
- Option RAM : 16b
- System RAM : 2KiB
Write to memory
Wrote and verified address 0x0801ee24 (100.00%) Done.

Starting execution at address 0x08002000... done.

MULTI-Module firmware flashed successfully.
```

benlye, 2021-01-29:
> MULTI-Module firmware flashed successfully.

Looks like it worked to me.  Are you saying there is still a problem?

c0debreaker, 2021-01-29:
> > MULTI-Module firmware flashed successfully.
> 
> The module was flashed correctly.
> 
> Your problem (in the picture) is that you are trying to use "Internal RF". Use "External RF"
oh, let me try now

c0debreaker, 2021-01-29:
I am seeing a new error when using external rf. `Module Status: Protocol Invalid` once I start changing the radio type to FrSky.

I'm pretty sure I used FrSky before when my JumperT16 was still on JumperTX.

benlye, 2021-01-29:
> I am seeing a new error when using external rf. `Module Status: Protocol Invalid` once I start changing the radio type to FrSky.
> 
> I'm pretty sure I used FrSky before when my JumperT16 was still on JumperTX.

Picture?

c0debreaker, 2021-01-29:
> > I am seeing a new error when using external rf. `Module Status: Protocol Invalid` once I start changing the radio type to FrSky.
> > I'm pretty sure I used FrSky before when my JumperT16 was still on JumperTX.
> 
> Picture?

https://i.imgur.com/xj54MuI.jpg

benlye, 2021-01-29:
Do any protocols work (show the module FW version in the status line)?
What are the module LEDs doing?

c0debreaker, 2021-01-29:
> Do any protocols work (show the module FW version in the status line)?
> What are the module LEDs doing?

That's the one I was wondering about. The leds on the modules aren't turning when connected to the Jumper T16. The led turns on when I connect usb cable. If you recall earlier, I was able to flash it too so it's working. And before I flashed OpenTX, I was playing with my Tinyhawk quad last week.

The other protocols are failing too. However, my crsf module for my other quadcopter doesn't have any issue.

c0debreaker, 2021-01-29:
I tried it again by changing MULTI back to OFF then back to MULTI, the leds turned ON. Then when I switch to the protocol, I got stuck at ASSAN. I can't scroll to choices anymore.

benlye, 2021-01-29:
Have you tried turning the radio off then on?  Don't plug the USB cable in as that will prevent the module starting up correctly.

What version of OpenTX have you installed?

c0debreaker, 2021-01-29:
Yes, I've tried rebooting T16 multiple times with the same behavior.

FW: opentx-t16
Version: is 2.3.11-TBS(31c20fc8)

c0debreaker, 2021-01-29:
My only last guess that maybe the culprit is the nightly version of opentx. I wanted crsfshot.

c0debreaker, 2021-01-29:
weird, it's working now. I see the version

benlye, 2021-01-29:
> weird, it's working now. I see the version

So all protocols are working correctly now?

c0debreaker, 2021-01-29:
So here's what I did. I set it to back to OFF, then select MULTI, then selected FrSky. When selecting FrSky, I have to scroll really fast. If I do it slowly, I'll get stuck at ASSAN. So I set to FrSky then D8. At that moment, the status was no multi telemetry detected. Then I saw Spectrum (INT) and (EXT). I chose (EXT) and I saw tons of bars. I covered the antenna with hand and I noticed the bar weakened. The leds at the back were turned on too. So I turned off T16 and turned it back on. Now the status shows the version :)

c0debreaker, 2021-01-29:
it's buggy but I guess because of the opentx bleeding version I selected which is why there are some bugs in the ui.

c0debreaker, 2021-01-29:
I was able to bind my Tinyhawk too! WOOHOO! :)

c0debreaker, 2021-01-29:
Thank you so much! The flash_multi helped a lot too!

breezerabbit, 2021-08-10:
mp\xepauhy5.mmc\MULTI-USB-Serial\installer_x64.exe is infected with Gen:Variant.Mikey.117370 and was moved to quarantine. It is recommended that you run a System Scan to make sure your system is clean.

pascallanger, 2021-08-10:
This is a false positive. Test it online with other antivirus software you'll see that it is clean...