pascallanger/DIY-Multiprotocol-TX-Module #510: DSMX RX Very slow packet rate

harry1453, 2021-01-31:
Trying to use a FSFly T-i8 with a JP5in1 for remote trainer. You get about 3 packets per second. It's a non-spektrum TX but works perfectly fine with a Lemon RX satellite receiver. 

LNA Enable / disable and increasing distance does nothing so I'm sure it's not signal swamping...

harry1453, 2021-02-01:
Trainer radio is configured as DSMX, 4 channels, 11ms packet time. Switching to 22ms didn't help... It's definitely transmitting because the multi gets *some* packets, just only a few.

harry1453, 2021-02-01:
Just played around with it some more. Turns out:

* You have to rebind for protocol settings to take effect (really obvious in hindsight but the radio didn't tell me that when I changed the settings)
* 11/22ms makes no difference
* **DSM2 works fine after a rebind.** DSMX doesn't work.

pascallanger, 2021-02-04:
I've just tried to bind with a Spektrum DX6 last gen using DSMX (it can't do DSM2) and everything is fine. Not even a packet loss between the 2 radios.
Since I can't reproduce your issue I have no idea on how to proceed.

harry1453, 2021-02-05:
Okay. Unless I investigate this further and find a cause I'll put it down to it being a clone transmitter for now.

closed 2021-02-05 by harry1453