pascallanger/DIY-Multiprotocol-TX-Module #513: S-FHSS no binding under v1.3.2.12

G-man177, 2021-02-02:
Hello,

i have flashed the new firmware on my modul and i can´t bind my XK110 with S-FHSS Protocol. I have tries the RF Freq. fine tune but without success. Under v.1.3.1.69 it works fine. Is something differrent?

Kind regads,
Gordon

pascallanger, 2021-02-04:
What are the LEDs doing on the module and the XK K110?
Do you have the module version displayed near the protocol?

closed 2021-02-08 by pascallanger

G-man177, 2021-02-08:
Hello,

unfortunately, I couldn't answer that quickly. My modul had a 128KB (STM32F103CB) mod.  Under v13.2.12 TAER it dosn´t displayed any modul version. The green LED on my modul are on and the red LED is flashed and after i push the bind-botton on the Xk110 it turns from red to off. iI expect a blue LED but it dosn´t com.  

Under version v.1.3.1.69 the XK110 LED turns to blue and version v.1.3.1.69 TAER is displayed.

Please look at my video link:   https://youtu.be/q8Uq3DViXT4  . 

kind regads,
Gordon

pascallanger, 2021-02-08:
Which module is it?
Which radio?

benlye, 2021-02-08:
You have replaced the original STM32 in your TMX5 module with a CB version?  Asking because as far as we know they **all** shipped with a C8 not a CB.

Be aware that the behaviour of the red LED is inverted on the TMX5 module.

G-man177, 2021-02-08:
I hade replaced STM32 ...C8 to a STM32F13CBT6. today i had flashed the firmware version multi-stm-serial-taer-v1.3.2.30.bin and everything works!

G-man177, 2021-02-08:
@pascallanger 

TMX5 with a STM32F13CBT6 mod and a Taranis X-Lite + adapter.

pascallanger, 2021-02-08:
The video shows a weird behavior. I think that you have a bad contact or something along this line.
Futaba S-FHSS is not a bind protocol meaning that there is no need to press bind. And if you press bind the module tells immediately the radio that bind is done. So from what I see there might be "some" communication time to time but poor enough to prevent things to work...

G-man177, 2021-02-08:
I dindn´t have a bad contact, it looks like a firmware issues.

G-man177, 2021-02-08:
Under under v1.3.2.12 is no binding possible. Instead of v.1.3.1.69 and v1.3.2.30 it is possible.

pascallanger, 2021-02-08:
Think what you want I tell you that your module does not operate normally. Go, fly and crash it's not my problem is you don't want to listen....

G-man177, 2021-02-08:
I think we're talking past each other or what shows me that my module is not working properly? Friends of mine me and i have flown under v.1.3.1.69 for a long time without failure or crash.

benlye, 2021-02-08:
> I think we're talking past each other or what shows me that my module is not working properly? Friends of mine me and i have flown under v.1.3.1.69 for a long time without failure or crash.

What Pascal is telling you is that there is no good reason why the firmware version you are reporting not to work should behave any differently to the versions that you are saying do work; this is an anomaly, and he suspects your module is faulty.