pascallanger/DIY-Multiprotocol-TX-Module #517: Radiolink R12DS, does it work?

dragonnn, 2021-02-05:
Hi,
I saw that recently multi-protocol module got support for RadioLink modules with made me really happy, but R12DS isn't listed. Did any one test it?
I am thinking about moving from my AT9 controller to RadioMaster T16S but I would prefer keeping my receiver at the current moment. I plan moving to ExpressLRS latter

pascallanger, 2021-02-05:
Check: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/401
Supported RX: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#RadioLink---74

closed 2021-02-05 by pascallanger

dragonnn, 2021-02-05:
I checked that first, just wanted to ask if maybe it is just not listed yet but maybe still "it could work".
Thanks anyway, so I need to replace my RX too when I get my new controller :/.