pascallanger/DIY-Multiprotocol-TX-Module #519: v1.3.2.12 Horus X 12S does not work with Multimodul Intern + Extern After reboot lost signal to Extern - All Receivers

Githubbachelor, 2021-02-06:
Hallo I have many Planes and other Models where I used years over a Multi Modul Jumper and the internel XJT FRSKY System from a Horus X 12S. Today I want to fly my ICON A5 with 13 Channel and I got bind my D8R II Plus with internal XjT D8 Mode FCC Non EU Open TX 2.3.11 an Multi Modul latest Versin Software. I want to bind my second Receiver a DSM2 but I cannot bind the Receiver with the Multi Modul Jumper. Than I tray another RX FLYSKY and also problem. No Receiver I can bind as second Receiver. But everything has worked before?!
If I use my external XJT Modul I can bind a second D8R II PLUs. And If I use my Radiomaster with second Multi Modul I can work with different Receiver. But with my Horus it does not longer Work.
Dear Pascal, can you please make the Multi Modul work with Horus 12S internal XJT D8R II Plus and extrenal Multi Modul Jumper again, so that I can bind the Receiver in that Modul as a second and give them Channel 9-17 in the second Multi Modul and 1-8 into the internal XJT Horus X 12S?
I have 8 Models which has worked before and now nothing??? What happen. Please help to reset the Multiprotokoll that it it work again?
Greetings Michael
Look pictures ...
[image]
[image]
[image]
[image]

I have many models with 2 receivers. The first is always FRSKY with Telemetry and the second Receiver is for all functions like light aso.

pascallanger, 2021-02-06:
You have posted your issue in 3 places now...
I have no idea what's the problem. I mean you are saying it was working before, before what? What have you done to break your setup?
You are staying that it works with a radiomaster so I don't the issue is the multi module.
On the horus, when you try to use the multi module what are the LEDs doing on the module? Do the LEDs change when you bind?
Lately referring the multi module is really easy, you just need to use flash multi and press the erase button...

Githubbachelor, 2021-02-06:
Yes I connected and posted it to Open TX. Andre told that that problem is no OPEN TX problem. If I use a extern XJT Modul and work with 2 D8R II Plus RX than the extern XJT is working with intern AND extern Modul same time. So it can only be a problem in the software anywhere. The Multimodul worked maybe 1 year ago. because I have always to receivers in my Water Planes and Speed ships

pascallanger, 2021-02-06:
This is a blank statement.
And by the way you are not answering any of my questions...
You just complain so far...

Githubbachelor, 2021-02-06:
All actual software and firmware is installed. Yes the extern LED is red flashing and I can only bind the extern Modul of the intern is off. I can bind the intern first than reboot the Horus. After switch intern to off. Than Choose extern and take Multi DSM 21F for Channel 9-14. After I can bind and try telemetry off. Than when is binded I switch intern on again to my RX D8R and it work with FRSKY. Than both are working. But If you switch off HORUS and on again the extern Modul does giv nothing out (NO LED RED) and the extern RX does not work only intern. If want to work I must do that procedure completely again but it can not be that that is a good result. If I want to use I need hours to open the model for taking receivers

pascallanger, 2021-02-06:
If there is no red led it means simply that the issue is in OpenTX while powering up the module end of it.

Githubbachelor, 2021-02-06:
The RED LED is working only when the extern Module is working alone or as I explain my procedure

closed 2021-02-06 by pascallanger

Githubbachelor, 2021-02-06:
I want to send you 2 pictures. How can I do?

kilrah, 2021-02-06:
You never mentioned the red LED not turning on in the issue you posted yesterday. Probably will be fixed by https://github.com/opentx/opentx/pull/8266