pascallanger/DIY-Multiprotocol-TX-Module #520: Traxxas Link 2218x

buzkill420, 2021-02-09:
I would like to contribute trying to reverse the protocol. Any help would be appreciated on going about creating dump files for the you.
Any resources available?  Thanks

buzkill420, 2021-02-09:
[image]

buzkill420, 2021-02-09:
[image]
I'm guessing transmitter isimportant as well!

pascallanger, 2021-02-10:
To make dumps, you need a logic analyzer (compatible Saleale 8CH cheap on ebay), solder it on the TX SPI bus (bus between the micro controller and the cyrf6936) to capture the signals clk, cs, mosi, and miso.
From there, we can try to understand the protocol.

pascallanger, 2021-04-18:
Will you provide dumps or should I close?

buzkill420, 2021-04-18:
I never got an analyzer... a bit pricey for me at the moment. You can close
and i can reopen if and when i get one.  Thanks

pascallanger, 2021-04-19:
A logic analyzer is $6 on ebay... But I'll go ahead and close it for now.

closed 2021-04-19 by pascallanger