pascallanger/DIY-Multiprotocol-TX-Module #521: NRF_CE pin function not defined

Rpi112, 2021-02-09:
Hi,

In Pins.h file, macros NRF_CE_on & NRF_CE_off are not defined unlike others (set with digital write function)..
Is there any reason ? The Pin number "NRF_CE_pin" is also not defined as PA13...

[image]

pascallanger, 2021-02-10:
I'm just not using it since the Atmega328 does not have it...

closed 2021-02-10 by pascallanger

Rpi112, 2021-02-10:
Thanks.. I am going ahead with defining these in my code as I am using STM32 board..
Hope this doesn't create any issues..

pascallanger, 2021-02-10:
I don't know the purpose but if you are designing a new protocol taking advantage of this pin then it won't be compatible with the atmega328 board...

Rpi112, 2021-02-10:
In NRF24l01_SPI.ino, the function "NRF24L01_SetTxRxMode" uses NRF_CE_off and NRF_CE_on for setting Rx and Tx mode. Hence I was wondering if any protocol using NRF will work if these macros are not defined?

pascallanger, 2021-02-10:
It's working so the answer is yes you don't need to play with CE to drive the NRF.

druckgott, 2025-02-09:
so if I have a STM32 I can leave the NRF_CE on the NRF24L01 free? Or do I have to connect it anywhere?

pascallanger, 2025-02-09:
NRF CE should be connected to VCC using a 1k resistor.