pascallanger/DIY-Multiprotocol-TX-Module #526: Follow up to "Follow up to "FrSky X9Lite with TBS MPM Module - No Telemetry" #491" #523

mikeller, 2021-02-10:
@tbs-trappy: It also has to be said that the solution presented in #523 and #525 was first suggested by me in https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/491#issuecomment-752057276.
This has also been edited out of #523 by @pascallanger.

pascallanger, 2021-02-10:
Please contact @mikeller for any future question as the subject expert on this matter.

mikeller, 2021-02-10:
@pascallanger: If you think that I am the subject expert on this, can I please get you to unlock the issues about this (#491, #523, #524, #525) and revert my original content so users seeking information can find it in a single place? Thanks.

kilrah, 2021-02-10:
"suggested by you" in a post where you're asking me to confirm you correctly understood what _I_ supposed? What the heck is wrong with you?

That was sorted in https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/491#issuecomment-770813371 and confirmed by multiple people already before you did, nothing more to add.

pascallanger, 2021-02-10:
Sorry but Issue #491 has all the information in one place for the users to to fix the issue if they need to. All other created issues are just repeat of an existing fix.

mikeller, 2021-02-10:
@kilrah: That was a theory that I'd had and had tried to verify before the exchange I had with you - this was just the first time that I'd voiced it, as my prior tests were not able to show this to be the case (see also https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/491#issuecomment-757075451).
Unfortunately as it turns out the multiprotocol firmware that I was using at the time of testing was not working properly as confirmed in https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/491#issuecomment-770417285, so my tests were bound to fail.
@pascallanger: The fix proposed in #491 is to short out the diode from the inverter output, and hence making it impossible to use the S.Port pin for communication from the TX to the module - I think there is value in raising awareness that is a better way to fix this by fixing the pull up, and retaining the bidirectional capabilites of the S.Port pin.

pascallanger, 2021-02-10:
@mikeller
And here we go for a heatted conversation again which is what I wanted to prevent!!!!
1. there has NEVER been a problem in any version of the firmware. TBS raised an issue which was in fact on their side trying to flash a version of the firmware expecting abootloader but without loading the bootloader first preventing it to work. This has been addressed with the Henry the main TBS dev and since then he has fixed his environment. **Bottom line you can try any version of the multi firmware and it will work**. You are mixing and making up things here...
2. I think you should read #491. I did post about replacing the pullup well before you mentionned anything. I've contacted TBS and the manufacturer before even posting on GitHub, due you want the emails with the time stamps? This is a mistake at the manufacture where somehow one of their engineer while revising the schematics has changed the value of the resistor on his own. The original schematics are showing the right value.

I don't even understand why we are talking and loosing time here for something which is already fixed, that's beyond my mind...

mikeller, 2021-02-10:
@pascallanger: I certainly don't think this needs to be heated - and my comments here were meant to be factual and not heated.

Re 1., in this case I must have misread @tbs-trappy's statement 'Tango2, with version beyond 1.3.1.59, does not work' (https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/491#issuecomment-765190691), and your response to it ('As you know the issue you raised was not a "real" issue since you were using a really old version of Flash Multi to update the firmware.', https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/491#issuecomment-770417285) as there being a problem with the version that I first installed on my module following the TBS instructions - apologies for that;

Re 2., I don't seem to be able to find your first comment mentioning a pull up before me, can you please supply the link to it? What I find a bit odd is that, even though you knew that the best fix was to change the pull up, you still suggested to users to fix this by shorting out the diode on TX, and hence make it impossible to use the port in bidirectional mode - to me this looked like giving bad advice to users, and I tried to clarify this.

As to why we are still talking about this, you seem to be unwilling to accept that others (me) proposed / pursued a solution that was different to the one proposed by you, and go to great lengths to censor and lock down this forum to prevent this from happening. This strikes me as wrong for a developer forum, and against the spirit and community rules in GitHub

pascallanger, 2021-02-10:
2. The exact same change that you are proposing is here posted 9 days ago (last comment on the thread): https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/491#issuecomment-770813371
Here is the picture with the resistor circled and changed by a 102=1K:
[image]

"you still suggested to users to fix this by shorting out the diode on TX, and hence make it impossible to use the port in bidirectional mode - to me this looked like giving bad advice to users, and I tried to clarify this."
Sorry but I don't think I'm giving bad advices to users since I've already submitted exactly what you are proposing 9 days ago on GitHub and the manufacturer... Your clarifications are even making things harder for the end  user since you keep posting new threads with a solution which is already there...

"As to why we are still talking about this, you seem to be unwilling to accept that others (me) proposed / pursued a solution that was different to the one proposed by you"
Sorry you are proposing the same just 9 days later... If you had new information with better results then everything will be perfectly fine and fully welcome!

"go to great lengths to censor and lock down this forum to prevent this from happening. This strikes me as wrong for a developer forum, and against the spirit and community rules in GitHub"
I'm doing this since you keep posting wrong information on a subject which is already closed since 9 days. If you had new information with better results then everything will be perfectly fine and fully welcome!

pascallanger, 2021-02-10:
FYI Shorting the diode was never a definitve solution but was instead a quick way to find the source of the issue since this is something that every one should be able to do easily. This is what I've asked you (and others) to test and what I've executed the same day I've received the X9-Lite to troubleshoot this issue. So yes I've posted pictures of the diode shortcut with a plier to show that it was the hardware issue I was thinking about. But then I've submitted the final solution after talking to the manufacturer to understand why the resistor value has been changed. What's wrong with that I can't understand... In less than 24 hours I had your issue solved, TBS informed, manufacturer informed, users informed and everything in one thread.

pascallanger, 2021-02-10:
I'm very sad about what's happening here since I spend a lot of time to deal with requests and reported issues in a timely manner.
This issue took at least 3 full days between the negotation with TBS to get the hardware, placing the orders, investigating and separating the different reports, reproducing, solving the different issues while trying to keep everyone informed.
And instead of a simple "thank you" when I've solved all the issues I'm getting all this?
I'm not paid for anything. This is all done on my part time during night and WEs sacrifying everything else. Surely my code is not perfect, my hardware designs are not perfect, I'm not perfect but I'm doing my best to have a working solution and I always spend the time to look into issues and drive them to conclusion.
I'm taking this really personnaly now and I'm considering to fully stop the development at this very moment.

mikeller, 2021-02-11:
@pascallanger: Thank you so much for starting this project, and for all the work you have put in and put in and are still putting in to get it to where it is now - this is a great achievement, and it is important to have an open source alternative in this space.
Apologies that I have not made this clear before.

mikeller, 2021-02-11:
Good point @pascallanger about finding a modification that is easy to do.
I have had another look and worked out how the pull up can be fixed without having to solder around small components at all, and by using a non-SMD resistor:

[image]

[image]

This should allow users with the affected module to fix this without reducing the functionality of the hardware even with moderate soldering skills (but make sure you insulate the resistor's legs so that they can not short against other components).

diego0815, 2021-02-12:
@mikeller like this you can lift up the 3v3 of the module, as you are taking the protective diode out of the equation.

I dont want to tell it won't work, but it's not sober engineering wise, as it opens up possibilities for Mr. Murphy.

mikeller, 2021-02-13:
@diego0815: It's been tested and proven to work, and it is a design that is used for bidirectional inverters in a lot of other places. The diode between S.Port and the inverter on the RX side is not really needed, and most designs for bidirectional inverters don't have it.

diego0815, 2021-02-13:
As I told, it will work under normal conditions, but don't plug it in wrongly.

Everyone is free to mod as desired, after all the value of a blown module won't financially ruin most of it's owners.

But that's personal taste, no need for further discussions.

mikeller, 2021-02-13:
@diego0815: Can you please explain what you mean by 'don't plug it in wrongly', and what you think the risks are coming from my proposal? This isn't really a matter of 'personal taste', as there either is a risk that can be explained, or it is just baseless fearmongering.

diego0815, 2021-02-13:
Ok @mikeller, if you look at the schematic you posted above, then think someone is connecting the battery to the sport pin, let's say 10V you will inject a couple of mA into the 3V3 rail, and thus raise the internal 3.3V, maybe above the voltage that's good for the rf module, but that I can't tell for sure, as I don't know the rest of this modules' schematic. Might be you have toast, might be not, so it's your decision if it's fear mongering or just precaution. 

As you stated it: "to me this looked like giving bad advice to users, and I tried to clarify this."

mikeller, 2021-02-13:
@diego0815: If somebody connected the S.Port pin to an over voltage as you say, this voltage will be fed into the output of the TX inverter through the diode on the lower leg, creating a short:

[image]

So if this happens then the inverter will be damaged at a lower voltage than what is required to feed more current through the 1K than the entire circuit consumes on 3.3V, and thus creating a problem.

So yes, you are right, feeding over voltage into the S.Port pin will damage the module. But this is the case with or without the modification postulated by me.

diego0815, 2021-02-13:
Probably you are right as the 74lvc inverter has a max Vo of 6.5V and a max Io source/sink capacity of 50mA. Still changing a CMOS logic IC means a simple repair, toasting the 4in1 RF module (the most sensitive component to over voltage) means a complete loss...

closed 2021-02-15 by pascallanger