pascallanger/DIY-Multiprotocol-TX-Module #527: X9D+ with STM32 + CC2500 DIY module No Telemetry status

kimman1, 2021-02-14:
Hi Pascal, 
My module still work perfectly fine before i update new firmware. 
With the version below 1.3.1.45 status in multi show the version but when i update to 1.3.2.30 it's show me no telemetry 
I've read the TBS hardware error and you said need to be change 10K resistor to 1K and with my DIY module. I replace 4.7K ohm resistor to 1K ohm resistor but the problem still there. 
[image]

pascallanger, 2021-02-15:
The resistor has nothing to do with your issue this is specific to the FrSky XLite 9 radio and TBS so a really special case...
What about the LEDs status? Do you still have control with your module, in other word is it only the telemetry? Do you have an inverter on your DIY module?
Have you upgraded the bootloader with the latest [Flash Multi v0.6.1](https://github.com/benlye/flash-multi)? Have you uploaded the latest firmware with Flash Multi v0.6.1?

kimman1, 2021-02-15:
I've fixed the issue. It's caused by STM32F103 with 64KB memory which you have mentioned before in another topic, i compile code and custom it myself and it's working now. 
But at this time i've a problem when i use betaflight lua scripts. It can not save the it's just read the value. I will have a video to describe more detail about this problem

pascallanger, 2021-02-15:
I know nothing about this script... You might want to work with betaflight instead.

closed 2021-02-15 by pascallanger