pascallanger/DIY-Multiprotocol-TX-Module #530: FRskyD / FRsky X : The module seems to reboot when throttled is max

grostim, 2021-02-17:
First time i use the Frsky protocol. I ha ve the same issue with FRskyD and FRskyX:
When throttle is maxed (beyond 90%-ish) the connection with the receiver is lost and the red LED blink. Once the throttle goes down, the connection restart.

I am not sure if this is an issue with my setup or the module itself.

Does this behaviour sounds familiar ?

benlye, 2021-02-17:
What's the receiver? Sounds similar to #505.

pascallanger, 2021-02-17:
Which radio?
Which module? If it has LEDs, what are the LEDs doing when it happens? What does the firmware version line states when it happens?

grostim, 2021-02-17:
@benlye  : you're right it looks very similar to #505

@pascallanger :
Radio: Turnigy 9xr with er9x firmware.
Module: iRangeX IRX4 Plus with firmware multi-stm-serial-aetr-v1.3.1.97.bin

receiver: All-in-one controller + receiver: CRAZYBEEF4FR

benlye, 2021-02-17:
Can you connect the board to Betaflight and see what's happening? The person who opened the other issue never came back to tell us, but my theory remains that it was a Betaflight fail safe.

pascallanger, 2021-02-17:
The module fw is pretty old but that shouldn't matter.
As far as I'm concerned it's a bug in the quad spi rx code, you should upgrade your quad to the latest version and it is still happening open an issue on their GitHub.

grostim, 2021-02-18:
> Can you connect the board to Betaflight and see what's happening? The person who opened the other issue never came back to tell us, but my theory remains that it was a Betaflight fail safe (due to the throttle value exceeding the failsafe threshold).

I will check. I am quite new to betaflight, but i will investigate.

> The module fw is pretty old but that shouldn't matter.
> As far as I'm concerned it's a bug in the quad spi rx code, you should upgrade your quad to the latest version and it is still happening open an issue on their GitHub.

You're most probably right . From what i can see the user from #505 is using the same controller board as me.

grostim, 2021-02-18:
So a few update:

I have tried to troubleshoot while connected to betaflight:

FRsky connection is reset when throttle reach 1880.
Until this value is reached everything is working ok. 
I don't think it is related to the failsafe setting as the failsafe should trigger only at 2150.
As soon as the throttle reach 1880, all channells fall back to their default values and the remote multiprotocol module has its red led blinking.
Connection is re-established as soon as the throttle is reduced for more than 2 seconds.

I have updated betaflight to the latest version, but unfortunately it does solve the issue.

pascallanger, 2021-02-18:
> As soon as the throttle reach 1880, all channells fall back to their default values and the remote multiprotocol module has its red led blinking.

red led blinking how?

grostim, 2021-02-18:
Hello again.
Here is a video showing the issue and the failsafe settings.
https://youtu.be/XsqCjT2cmbo
I think the blink is a fast blink .

I will share separate pictures of the radio protocol settings.

benlye, 2021-02-18:
What's on AUX12 and why is it increasing when you increase throttle, but bouncing?  Could it be spiking above the failsafe threshold?

pascallanger, 2021-02-18:
Autobind is set to yes. My guess is that you have a mixer on ch16 which makes the bind to happen when you increase throttle...

closed 2021-02-18 by pascallanger