pascallanger/DIY-Multiprotocol-TX-Module #531: copy other modules UID / Set UID using transmitter

ltwin8, 2021-02-21:
Hello, I have bought an jumper t-lite a few days ago and was wondering if it is possible to migrate seamlessly between two transmitters, I have a Qx7 as my main Radio and already got it to work with the frisky stuff using CloneTX Subprotocol, nice work on that, it is really a perfect feature!, I was thinking if it would be possible to read out the UID of my Taranis' 4-in-1 module an copy it to the jumper module so I can seamlessly fly all models on the Qx7 with the jumper without rebinding. Maybe it would be possible to have a Lua Application running on the TX that would read the module settings (UID, Powervalues of PA (I guess there are not only full and low power), maybe if possible SWR) ad maybe also let it make changes to the module so for example:

I have an OMP M2 and would like to fly on both radios. Sure one could recompile the firmware every time and force an ID, but that would mean to set an known ID on both and rebind everything, quite a bit of work. 

And also to be honest I would think that recompiling every time maybe some user errors will find its ways into the firmware when wrong configured, also I have not managed to get a *.bin file from ArduinoIDE to flash the t-lite's module so it would be great to have a Lua script to do this on the fly and let the firmware be generic like it is now.

I would be happy to help implement this, but I am not familiar with LUA, I know a bit C++ but I am more into hardware

pascallanger, 2021-02-21:
Compile your own versio with FORCE_GLOBAL_ID defined in the _config.h file. You can read your module ID using Flash Multi.

closed 2021-02-21 by pascallanger

ltwin8, 2021-02-21:
I have wrote, I can not compile it and export it as binary.

is there any way to change the ID without recompile?

Or something like cloneTX for the other protocols like OMP, DSM, FlySky