pascallanger/DIY-Multiprotocol-TX-Module #534: surface receiver DSMR protocol missing

ralfthiele, 2021-03-01:
I have a couple of these Spektrum SR215-SR515 surface receivers. They are tiny and long range and perfect for small applications. Unfortunately I am not able to bind them to the multi protocol module. The protocol is DSMR and cannot be bind in any of the DSM protocols available for the Multi module. Spektrum claim that they are downwards compatible with DSM2 protocol but even  Spektrum Airware Transmitter not bind them. Is it possible to include the DSMR protocol in the firmware, that those usefull receivers can be bound to the multi module? Would be great.

pascallanger, 2021-03-01:
For this I would need access to at least one original TX and RX. If you are willing to ship then PM me on RCGroups pseudo hpnuts.

pascallanger, 2021-03-30:
The good news:  I have DSMR working at 11ms. I can control the SR215 I have which has only 2 channels and even get telemetry back from it. I don't know how a RX with more outputs would behave but the protocol can transmit up to 7 channels so I'm thinking it should just work.

The bad news:
 - The 23 RF channels depend on the ID (like DSMX). They've updated the pseudo generator and despite many trials I haven't been able to figure it out.
 - Need to see if I can implement the 5.5ms version.

@benlye @richardclli If you want to have a look at the pseudo generator you are welcome!

pascallanger, 2021-03-30:
@ralfthiele Can you test v1.3.2.66 available here: https://downloads.multi-module.org/latest-test/

ralfthiele, 2021-03-31:
tested v1.3.2.67 and it binds perfect with a SR515 DSMR receiver, running all 5 channels in setup order.
I will do the range test with a race boat on coming weekend..

closed 2021-04-18 by pascallanger