pascallanger/DIY-Multiprotocol-TX-Module #542: Bind with new Cheerson CX10 from 2020 fails

marph43, 2021-03-13:
Dear all
I recently purchased a new Cheerson CX10 and tried binding it with the Multiprotocol-TX-Module of my Jumper T12. I tried every CX10 model setting I could find, but so far no luck with binding. Is it possible that they changed the protocol again or am I doing something wrong? The PCB is blue and has "WOJ-MINI-10RX" and the date 05/28/2020 written on it. I updated the stock firmware of the Multi-Module to the following version: multi-stm-serial-aetr-v1.3.2.30.

pascallanger, 2021-03-13:
Let's try first to bind your CX10 with an old version of the FW like: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/releases/tag/v1.3.1.78

marph43, 2021-03-13:
Thanks for the quick answer. I just tried your proposal. But no luck either (at least not with the "Blue PCB" setting).

pascallanger, 2021-03-13:
Ok your TX has an external module or internal?

marph43, 2021-03-13:
External. JP4IN1-SE

marph43, 2021-03-20:
Can I provide any help or additional information to get this to work? Is the cause of the problem maybe the version of my OpenTX firmware? Or do I maybe not to reset the EEPROM?
Here is what happens when I start the bind process: It keeps on beeping (beep-beep, pause, beep-beep, pause etc.)

jxweric, 2021-12-02:
Hi, I'm using the Cheerson CX-10 with blue PCB for a school project. We are designing our own controller. We are transmitting the bind signal with a nRF24L01, but failing to bind to the drone (similar problem as described above), can you please help me with it? Let me know if you need more information from me!

pascallanger, 2021-12-06:
Hi, without having access to the TX and quad this is going to be difficult. If you send me one or sponsor me to purchase one I'll have a look. Pascal

vyoufinder1, 2023-03-04:
I'm having the same problem.  My orange CX10 won't bind.  Any updates?

pascallanger, 2025-07-26:
This protocol is a variation of the SG-F22 protocol.
Similar bind/normal packets structure. I haven't checked the normal RF frequencies but I guess the same calculation will apply.
Will you take a go at adding a new subprotocol to the SGF22 protocol or should I?

pascallanger, 2025-07-27:
For the RF frequencies, can you try to see if they obey the same %5 rule? If it is then you already have 2 sets of frequencies on the 5 and finding the other sets is not complicated.

hexwell, 2025-07-27:
I tried but I think they do not.

Here is the counterexample:

For original TX1 ID values `{ 0x4C, 0xD7 }`, `val` comes out as `0`.

Then I tried ID values `{ 0xDF, 0x0C }`, for which `val` also comes out as `0`.

However the drone does not respond to the same frequencies (`{ 55, 66, 71, 60 }`).

I also tried with `% 4` instead of `% 5` and without `val--` if `id[0] > (0xFF - id[1])`.

`%4` counterexample: ID = `{ 0x81, 0x2A }`
no `val--` counterexample: ID = `{ 0x56, 0xEB }`
`%4` and no `val--` counterexample: ID = `{ 0x84, 0xAF }`

pascallanger, 2025-07-27:
You are doing too big jumps on the values... You don't know the effect of such jumps. Just set the RF frequencies of one of the ID and then increment only the low part of the ID one by one. You can do this increment without reuploading the code by adding rx_tx_addr[3] with RX_num. Doing so you can use the receiver number on your radio to test the different values.

hexwell, 2025-07-27:
My idea was that if this drone uses a similar table (with `index = ID % 5`), then all the IDs which give the same result `% 5` should use the same frequencies, but I found those counterexamples that disprove the premise.

I chose those values simply by calculating all the values which `% 5` are equal to one of the IDs I have already, choosing one at random, and testing if it worked with the same known frequencies.

<br>

But anyways, I get your point that it's best to test small changes and find working examples to look for patterns.
If it is `% n` and I fix the frequencies, then after `n` increments of the ID it should bind again.

Thanks for the `RX_num` tip, it greatly sped up testing!
I set constant frequencies and used `packet[3] = rx_tx_addr[3] + RX_num` and tested all the receiver numbers from `0` to `63`, only `0` worked, no luck here.
I also tried with `packet[2] = rx_tx_addr[2] + RX_num` in case it's little endian, `0 - 63` only `0` worked.

Earlier I found some new IDs which were working with known frequencies, let me find them again and I'll comment them here

pascallanger, 2025-07-28:
New code version with the SGF22/CX10 subprotocol pushed.
Doc: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#sub_protocol-cx10

I've implemented everything except packet[1]=0x11 during bind since I'm not sure if it's really needed...

For the ID/frequencies, I think we should look at it the same way as I'm doing it usually which is:
1.  set an ID in the code, can start with 00 00
2.  bind
3.  use the option field to find the 4 RF frequencies
4.  increment the ID by using RX_num and restart to step 2

To get the option field to appear on the radio menu within the module parameters (protocol, subprotocol...) you need to modify OPTION_NONE to OPTION_OPTION on this line: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/5ce6c6f544729d2a9d8dc91970b971b1918cf315/Multiprotocol/Multi_Protos.ino#L477

Then use the variable `option`  to set the RF frequency. On the radio you don't need to validate the option field for every value change since it will be applied on the fly. Change this line: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/5ce6c6f544729d2a9d8dc91970b971b1918cf315/Multiprotocol/SGF22_nrf24l01.ino#L75
By this: `XN297_RFChannel(option);`