pascallanger/DIY-Multiprotocol-TX-Module #544: Why internal TX16S module is not able to bind AFHDS but external (iRangeX) does?

prikrylm, 2021-03-16:
Tested on TX16S, with external module iRangeX. The same radio, the same TX module firmware, the same binding process. But internal module is unable to bind AFHDS. Tried on FS-R9B and FS-GR3E.
Where am I making mistake?

pascallanger, 2021-04-09:
It sounds really strange. Unless the A7105 is defective on the internal module?
The Flysky protocol is there since years and I'm using it frequently to fly indoor...

prikrylm, 2021-04-09:
Yes it is strange. In addition binding AFHDS 2A works without problems on both TX modules.
Is there a way to check if the TX chip is defective? 

Otherwise, this "problem" has no priority. I use AFHDS very rarely.

prikrylm, 2021-04-29:
Definitely some hardware problem. Another TX16S with the same TX module firmware is able to bind AFHDS receivers.

closed 2021-04-29 by prikrylm