pascallanger/DIY-Multiprotocol-TX-Module #545: Telemetry and false warnings.

ASKFan, 2021-03-17:
Hello everybody,

I use the TX16S from Radiomaster with the DSMX protocol.
The problem here is that the complete telemetry leaves at approx. 200-250 meters. I asked Radiomaster support about this.
The support said it is due to the transmission of DSMX, the telemetry does not go that far. I do not believe this statement so much. Because if I use the same receiver in connection with the Spektrum DX7-G2, then I have complete telemetry up to 1000 meters.

There is also the problem that at a distance of 50 meters on the ground, warnings such as "weak signal or critical signal" appear. These warnings are false. I tested the range of the TX16S up to 1200 meters. It's all ok. Is there a solution to this?

greeting
Dietmar

kilrah, 2021-03-17:
No idea without you mentioning the receiver model. If it's a "short range telemetry" one then the RM answer is correct.

ASKFan, 2021-03-17:
Said receiver is the Lemon RX 0051U 7-channel full-range telemetry
Tas Prolem also exists with the spectrum AR6600T, AR8000T., AR8010T
All receivers have the latest firmware.
As written, with the transmitter from Spektrum it works without any problems. Not with the Radiomaster.

pascallanger, 2021-03-17:
"There is also the problem that at a distance of 50 meters on the ground, warnings such as "weak signal or critical signal" appear. These warnings are false. I tested the range of the TX16S up to 1200 meters. It's all ok. Is there a solution to this?"
Most DSM receivers do not send RSSI information. Therefore OpenTX is using instead the RSSI information from how well the module receives the telemetry for the weak and critical warnings. Additionnaly, OpenTX default levels are for FrSky and not for DSM so some values adjustments are necessary. But again and as you it has nothing to do on how well the RX is receiving the TX.
Feel free to change the source of the RSSI if your RX does send this info.

"The problem here is that the complete telemetry leaves at approx. 200-250 meters."
I'm wondering if we are comparing the same thing here. OpenTX will complain if it loses telemetry for a short period of time. As far as I know a Spektrum radio won't compalin and will keep the last value until it receives a new one.
So my question is: do you fully loose telemetry or you get instead messages "telemetry lost/recovered" above 200-250m?

ASKFan, 2021-03-17:
The telemetry is not only gone for a moment, it's gone completely.
RSSI / VSpeed ​​/ ALT ............ everything is completely lost.

At a distance of 50 meters I have full telemetry, but here comes "weak signal" that is not really correct.

pascallanger, 2021-03-17:
> At a distance of 50 meters I have full telemetry, but here comes "weak signal" that is not really correct.

I've already explained the why above and what you can do.

> The telemetry is not only gone for a moment, it's gone completely.

Compile the code with #define USE_CYRF6936_CH15_TUNING uncommented in _config.h. Assign a pot on CH15 and see if it will have an effect on the distance. Report your findings.

pascallanger, 2021-04-09:
No answer from my last post... Closing this thread.

closed 2021-04-09 by pascallanger