pascallanger/DIY-Multiprotocol-TX-Module #547: Support for Eachine E016F

marph43, 2021-03-20:
Dear all
I recently purchased an Eachine E016F and tried binding it with the Multiprotocol-TX-Module of my Jumper T12. I know that currently there is only support for E016H, but I tried my luck anyways. I tested the E016H and other protocols which were used by Eachine, but none worked. I tried firmware multi-stm-serial-aetr-v1.3.2.30 and multi-stm-serial-aetr-v1.3.1.78.
Is it possible to implement this protocol as well? I would be very happy to help by providing information and testing.

pascallanger, 2021-03-27:
Have you tested the E016Hv2 protocol?

marph43, 2021-03-28:
Yes. I've tested both with the newest firmware multi-stm-serial-aetr-v1.3.2.61, but no luck. E016Hv2 provides an option for frequency tuning. Is it possible that it would bind if I set a certain value other than 0 there? If so, what do you suggest I should try?

255h, 2021-03-30:
Hey there.
Same issue here. 

Tied every combination possible in last multiprotocol firmware.
I believe it's because this model uses different radio chip than previous models from Eachine.

pascallanger, 2021-03-30:
I would need access to the TX and quad to add the protocol (if doable), 2 solutions:
 - you send me the hardware
 - you participate to the purchase, the E016F is currenlty at 24.29€ on Banggood 1 week delivery

marph43, 2021-04-02:
It would be great if you could implement this protocol. I would like to contribute, so you can purchase the hardware. Do you have Paypal?

pascallanger, 2021-04-02:
Yes I do PM me on RCGroups user:hpnuts .

marph43, 2021-04-03:
Ok. I tried to send you a message via RCGroups. I did not know this site before, so I am not sure if it was successfull. Did you receive anything?

pascallanger, 2021-04-09:
BG has still not sent it... I'll let you know the progress.

pascallanger, 2021-04-17:
I'm sorry but BG has still not sent the quad... I'm wondering if they are out of stock. I've sent them an inquiry email.

pascallanger, 2021-04-24:
I've finally cancelled my purchase with Banggood since the E016F has still not been sent after a month! I think the web site does not show the stock level and they keep you waiting... I've placed the order on Amazon with a next day delivery so hopefully I will have it tomorrow to look at, it's more expensive (+10€) but you have it right away 😃 . More news in the coming days.

pascallanger, 2021-04-26:
The RF component is a FC298TX like for the E016Hv2 protocol: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/469
It's using GFSK @250Kbps but the bind payload is long (39/40 bytes) which likely indicate that FEC is enabled but what kind...
The packet structure is also different which contributes to the fact that more options have been activated.
I haven't been able to find a doc on this chip so we don't know what it's doing.

Here are a couple of binds, Bind0 is 1 bind then I power cycled and took 4 consecutive bind packets, as you can see Bind0 has different values so the bind is different at each boot:
  | Bind0 | Bind1 | Bind2 | Bind3 | Bind4
-- | -- | -- | -- | -- | --
ID | ED | ED | ED | ED | ED
ID | 27 | 27 | 27 | 27 | 27
ID | 17 | 17 | 17 | 17 | 17
ID | 94 | 94 | 94 | 94 | 94
ID | 61 | 61 | 61 | 61 | 61
  | 5 | 05 | 05 | 05 | 05
PID | **4**6 | **5**6 | **6**6 | **7**6 | **4**6
  | 41 | 41 | 41 | 41 | 41
  | 0B | 0B | 0B | 0B | 0B
  | 25 | 25 | 25 | 25 | 25
  | 75 | 75 | 75 | 75 | 75
  | BE | 8B | 8B | 8B | 8B
  | 75 | 75 | 75 | 75 | 75
  | 00 | 00 | 00 | 00 | 00
  | 2A | 2A | 2A | 2A | 2A
  | A4 | A4 | A4 | A4 | A4
  | 94 | 94 | 94 | 94 | 94
  | A4 | A4 | A4 | A4 | A4
  | 6F | 6F | 6F | 6F | 6F
  | 26 | 26 | 26 | 26 | 26
  | 1A | 1A | 1A | 1A | 1A
  | 50 | 50 | 50 | 50 | 50
  | E7 | E7 | E7 | E7 | E7
  | AF | AF | AF | AF | AF
diff  | **29** | 2F | 2F | 2F | 2F
diff  | **8**E | 3E | 3E | 3E | 3E
diff  | 5**8** | 5E | 5E | 5E | 5E
diff  | **F**E | 7E | 7E | 7E | 7E
diff  | **CF** | A5 | A5 | A5 | A5
  | EF | EF | EF | EF | EF
  | 4E | 4E | 4E | 4E | 4E
  | B8 | B8 | B8 | B8 | B8
  | CA | CA | CA | CA | CA
  | 33 | 33 | 33 | 33 | 33
  | A1 | A1 | A1 | A1 | A1
diff  | 6**2** | 66 | 66 | 66 | 66
diff  | **DA** | 86 | 86 | 86 | 86
CRC? | **DC** | 41 | 41 | 41 | 41
CRC? | **6B** | 0A | 0A | 0A | 0A
part of packet?  |   |   | 18 | 18 | 18

Looking at it I'm wondering if the CRC could be in fact spread on 4 bytes 66 86 41 0A which could be the effect of the FEC.

pascallanger, 2021-04-26:
Normal packet with Bind1 values.
5 RF channels: 45, 62, 74, 71, 54, full loop in 62875µs, change every 12575µs, repeat 5 times per channel every 2515µs.

Note that the payload might be shifted by up to 7 bits...

  | Bind6 | Normal
-- | -- | --
ID | ED | ED
ID | 27 | 27
ID | 17 | 17
ID | 94 | 94
ID | 61 | 61
  | 05 | 05
PID | 76 | 46
  | 41 | 41
  | 0B | 0E
  | 25 | 28
  | 75 | 4C
diff | 8B | 6E
  | 75 | 75
  | 00 | 00
  | 2A | 2A
  | A4 | A4
  | 94 | 94
  | A4 | A4
  | 6F | 6F
  | 26 | 26
  | 1A | 1A
  | 50 | 50
  | E7 | E7
  | AF | AF
diff | 2F | 1C
diff | 3E | 91
diff | 5E | AE
diff | 7E | BE
diff | A5 | 01
diff | EF | 6F
  | 4E | 4E
diff | B8 | 88
diff | CA | EE
diff | 33 | 4C
diff | A1 | 98
diff | 66 | E6
diff | 86 | 09
CRC? | 41 | 60
CRC? | 0A | 25
part of packet? | 18 | 20

pascallanger, 2021-04-28:
I'm about to give up... I don't know how the FEC has been implemented. Since the original message is unknown it's quite hard to get any grasp on it.
Usually the feature like flip, mode, rate, etc are changing a single bit so I dumped those. The table below shows the comparaison between no flag and the flag active the 3rd line value is a xor which shows which bits have changed (0->1 or 1->0):

None | 49 | 4A | 46 | F2 | 61 | A5 | 0E | 7A |   | 01001001 | 01001010 | 01000110 | 11110010 | 01100001 | 10100101 | 00001110 | 01111010
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
Flight mode | 49 | 4A | 46 | F2 | 61 | E0 | 8E | 7A |   | 01001001 | 01001010 | 01000110 | 11110010 | 01100001 | 11100000 | 10001110 | 01111010
  |   |   |   |   |   |   |   |   |   | 00000000 | 00000000 | 00000000 | 00000000 | 00000000 | 01000101 | 10000000 | 00000000
  |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
None | 49 | 4A | 46 | F2 | 61 | A5 | 0E | 7A |   | 01001001 | 01001010 | 01000110 | 11110010 | 01100001 | 10100101 | 00001110 | 01111010
Headless | 49 | 4A | 46 | B7 | E1 | A5 | 0E | 7A |   | 01001001 | 01001010 | 01000110 | 10110111 | 11100001 | 10100101 | 00001110 | 01111010
  |   |   |   |   |   |   |   |   |   | 00000000 | 00000000 | 00000000 | 01000101 | 10000000 | 00000000 | 00000000 | 00000000
  |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
None | 49 | 4A | 46 | F2 | 61 | A5 | 0E | 7A |   | 01001001 | 01001010 | 01000110 | 11110010 | 01100001 | 10100101 | 00001110 | 01111010
RTH | 49 | 4A | 46 | F2 | 69 | AB | 0E | 7A |   | 01001001 | 01001010 | 01000110 | 11110010 | 01101001 | 10101011 | 00001110 | 01111010
  |   |   |   |   |   |   |   |   |   | 00000000 | 00000000 | 00000000 | 00000000 | 00001000 | 00001110 | 00000000 | 00000000
  |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
None | 49 | 4A | 46 | F2 | 61 | A5 | 0E | 7A |   | 01001001 | 01001010 | 01000110 | 11110010 | 01100001 | 10100101 | 00001110 | 01111010
Speed 1 | 49 | 4C | F6 | F4 | E1 | A5 | 0E | 7A |   | 01001001 | 01001100 | 11110110 | 11110100 | 11100001 | 10100101 | 00001110 | 01111010
  |   |   |   |   |   |   |   |   |   | 00000000 | 00000110 | 10110000 | 00000110 | 10000000 | 00000000 | 00000000 | 00000000
  |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
None | 49 | 4A | 46 | F2 | 61 | A5 | 0E | 7A |   | 01001001 | 01001010 | 01000110 | 11110010 | 01100001 | 10100101 | 00001110 | 01111010
Speed2 | 49 | 4A | 4E | F1 | 21 | A5 | 0E | 7A |   | 01001001 | 01001010 | 01001110 | 11110001 | 00100001 | 10100101 | 00001110 | 01111010
  |   |   |   |   |   |   |   |   |   | 00000000 | 00000000 | 00001000 | 00000011 | 01000000 | 00000000 | 00000000 | 00000000
  |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
None | 49 | 4A | 46 | F2 | 61 | A5 | 0E | 7A |   | 01001001 | 01001010 | 01000110 | 11110010 | 01100001 | 10100101 | 00001110 | 01111010
Flip | 49 | 4A | 47 | 75 | E1 | A5 | 0E | 7A |   | 01001001 | 01001010 | 01000111 | 01110101 | 11100001 | 10100101 | 00001110 | 01111010
  |   |   |   |   |   |   |   |   |   | 00000000 | 00000000 | 00000001 | 10000111 | 10000000 | 00000000 | 00000000 | 00000000

Flight and headless are interesting since the same bit positions have flipped in different bytes. But for the other entries nothing is jumping out...

marph43, 2021-04-28:
Ok. Thanks a lot for your work so far!

So if you are talking about these flags, the binding process did already work? Just the special functions don't?

pascallanger, 2021-04-28:
Nothing works, if I can't figure how to decode/code the payload there is nothing...

pascallanger, 2021-05-01:
After a few nights spent on it, I've been able to find something.
The 8 bits of a payload byte is spread over 18 bits using this:
b0\|b0\|-\|b0\|-\|b0\|b0\|b1 | b2\|b3\|b4\|b4\|b5\|b6\|b7\|-\|b0^b2^b4^b5^b7\|b0^b1^b2^b3^b4^b6^b7\|b1^b2^b3^b4^b5^b7\|b0^b3^b4^b5^b6^b7\|b1^b6^b7\|
The next byte has the same applied, bitshifted by 15 and most likely xored with the previous result mixing the bits with the prev value adding to the complexity...
The problem is I can't go back to the original payload since after applying the FEC above the payload is scrambled/xored.
While I can replace some parts of the payload like throttle, aileron, elevator, rudder, trims, there is no way to calculate the associated CRC...

So basically end of story, no way to reverse this protocol further without the scramble pattern...

ac81, 2021-06-26:
Just a note for anybody searching, that JJRC H94 appears to be the same as E016F.

LilTeo14, 2021-08-17:
As note also it looks than the WLToys F949G or P707G appears to be the same as E016F

marph43, 2022-04-20:
I just found out that the JJRC H36F is almost exactely the same when it comes to features as the Eachine E016F. JJRC H36F is supported by DIY-Multiprotocol. I tried bindng with this protocol but it didn't work. Is it possible that the protocol of E016F is more similar to E010R5 (H36F) than to E016H or E016HV2? If so, could this help to get support for E016F?

pascallanger, 2022-04-21:
No...