pascallanger/DIY-Multiprotocol-TX-Module #548: Generic Max Throttle Function

richardclli, 2021-03-22:
Last week when my friend tries the WFLY2 protocol on his FPV quad runner and figures out problems.  In betaflight settings, the requirements of the throttle range is 1000-2000us.  In which the range of WFLY2 protocol only have 1102-1941us in output range.  I noticed that DSM has this problem as well, therefore when using DSM with FPV runners, you need to select the Max Thr. settings.

So I am thinking, because of the requirement of the throttle range for FPV runners, a more generic function for ensuring the extended output range maybe required for all protocols that is used by general purpose receivers.

Any comments?

Tygrys-1, 2021-03-22:
The idea looks ok. OTOH you can tune betaflight to scale RC channels to reuired values by rxrange cli parameter.

richardclli, 2021-03-23:
Just figure out another way to achieve this and no need to change the code.  Just enable the extended limits in OpenTX and tune the output to 150%, this can cap the output of the WFLY2 protocol to 1000-2000us.

closed 2021-03-23 by richardclli