pascallanger/DIY-Multiprotocol-TX-Module #551: FrSKY Xlite Pro no longer sees IRangeX IRX4 Lite after firmware update

Chris15X, 2021-03-29:
Hi,

I upgraded my FrSKY Xlite Pro and now my multiprotocol module is no longer accessible from the radio. The radio no longer sees it. These are the firmware version I am using:

FrSKY Xlite Pro is now running 2.3.12
The ISRM-Pro is now on 1.1.0 / 2.1.6 FCC
The multiprotocol module (IRangeX IRX4 Lite) is showing Serial Port DFU Device in Flash Multi v0.6.1. The Firmware shows 1.3.2.61 in Flash Multi. I also tried 1.3.2.58, 1.3.2.30, 1.3.2.12 and that didn't work either.

I am not sure what to try next. The multiprotocol module worked great before it was flashed.

Thank you for the help.

kilrah, 2021-03-29:
You already posted somewhere about that and got an answer.

Chris15X, 2021-03-29:
Hi, 

I reposted because Ben said I posted it in the wrong place and that it likely has nothing to do with Flash Multi, but it probably has to do with the Firmware of the Multiprotocol module. My issue has not been resolved yet. 

Thank you

pascallanger, 2021-03-29:
What does "no longer accessible" means?
Have you tried downgrading back to the previous version since you state that the old version was working?
What are the LEDs doing on the module?
Do you have control and you are only missing the module version?

Chris15X, 2021-03-29:
I previously had upgraded using Arduino and that was what I tried first. Unfortunately I made the mistake of flashing before trying to figure out what version it had. 
When I go to select a Model the External RF Mode no longer show Multi as an option. 
When the radio is on the Multimodule LED is dark.

pascallanger, 2021-03-29:
"When I go to select a Model the External RF Mode no longer show Multi as an option."
Wrong OpenTX firmware...

Chris15X, 2021-03-29:
I downloaded the latest version from Frsky. https://www.frsky-rc.com/taranis-x-lite-pro/ 
Should I have downloaded it from OpenTX instead or maybe I should try a different version?

pascallanger, 2021-03-29:
Do you really think FrSky is going to give out a firmware which supports the Multi module?

kilrah, 2021-03-29:
Again you got the answer already on your RCG post, FrSky builds don't include multi or CRSF, only their own products and the generic choices.

pascallanger, 2021-03-29:
Since things are being discussed elsewhere and obviously it's not a Multi firmware issue I'm closing this thread.

closed 2021-03-29 by pascallanger