pascallanger/DIY-Multiprotocol-TX-Module #552: Can’t bind to FrSky R-XSR F.Port version

aa-calvo, 2021-04-05:
Hi,

I’m not being able to bind my R-XSR receiver, running ACCST D16 v2.1.1 (F.Port firmware update only), to my Jumper T18 radio, with the module running Multiprotocol v1.3.2.61. 

I know there are some old issues about incompatible firmware versions, but I don’t know which firmware version do work and where to download it from. Also, I would prefer it the firmware supported the F.Port protocol for telemetry.

Any help is appreciated, thanks!

aa-calvo, 2021-04-06:
It works with RXSR-FPORT_LBT_ACCST_191128.frk firmware.

It can be downloaded from FrSky download page: https://www.frsky-rc.com/r-xsr/. Where it says “ History firmware DOWNLOAD (not compatible with v2.x.x)”

closed 2021-04-06 by aa-calvo

HazzaHFPV, 2021-04-06:
It also works with ACCST v2.1.0 when you choose FrSkyX2 Protocol in the Radio. There is no difference in telemetry between FPort and "normal" S.port Firmware. That ist only a difference in communication between receiver and flight Controller.

aa-calvo, 2021-04-07:
I tried using FrSkyX2 and I was able to bind successfully, but I think there is a problem with the F.Port telemetry on version 2.X.X. 

I don’t think it’s related to the multi protocol transmitter because I observed the same problem on a Taranis Q7. It could be either Arducopter or the telemetry script.