pascallanger/DIY-Multiprotocol-TX-Module #554: No binding of Hott possible and no more telemetry.

Basti77, 2021-04-12:
Hello, 
today I noticed that probably pages of the last update to OTX 2.3.11
When binding Hott is no longer possible. 
I still had an old version on the multimodul irx4 lite for my Xlite 1.3.0.xx.
Since the transmitter asked me to update I installed the current version.
Multi-stm-serial-taer.v13.2.61.bin
Now the transmitter doesn't complain anymore, but binding is still not possible.
Additionally the telemetry has disappeared. 
Does anyone have an idea?

pascallanger, 2021-04-12:
Frequency tuning, HoTT requires a positive value, try +40 to bind.

Basti77, 2021-04-12:
Excuse me, but how do you usually do that? 
Or what should I do with this info?

kilrah, 2021-04-12:
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Frequency_Tuning.md

pascallanger, 2021-04-12:
By reading the doc maybe: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#hott---57

Basti77, 2021-04-12:
The hint was in the text, but in a different context. The FW of the GR-16 was once again too old.
Update Via the Gr Studio and bind + telemetry works.
Thanks for the food for thought.
(Crap now, I have to remove the receiver again for any Plane....the reason for flash laziness).
Ticket can be closed

closed 2021-04-12 by pascallanger