pascallanger/DIY-Multiprotocol-TX-Module #555: How did you get the protocols?

ErfolgreichCharismatisch, 2021-04-14:
How did you get the protocols for the drones? Reverse engineering?
I am trying to reverse engineer the A20 of Potensic.
Can you share the commands for it - probably using UDP - or how to reverse engineer it?

pascallanger, 2021-04-15:
The potensic A20 protocol is already available: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#Potensic---51
Yes reverse engineering.

ErfolgreichCharismatisch, 2021-04-15:
Can you explain how to use it with python? so without your hardware, just a wifi module plus python.

Also, I am curious how you reverse engineered the protocol.

pascallanger, 2021-04-15:
That's not how it works... If it was that simple... A wifi module won't be able to be used here.
You need a SDR.

ErfolgreichCharismatisch, 2021-04-15:
In this case it is, because the drone can be controlled via smartphone app, which connects via wifi.
Who did the reverse engineering and can tell me a thing or two?

ErfolgreichCharismatisch, 2021-04-15:
Can you help me out?

pascallanger, 2021-04-15:
You need to understand the content of the packets. Move throttle and look in the packets what's changing, stake thing for all other channels and features. One you have identified most of the bus then you can reproduce what's being sent.

closed 2021-04-15 by pascallanger

ErfolgreichCharismatisch, 2021-04-15:
How do I convert these packets into python code?