pascallanger/DIY-Multiprotocol-TX-Module #557: New Protocol Kyosho KT-531P

AndreasAusD, 2021-04-16:
based on this [thread](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/379) which is for the KT-19 transmitter and on this [Radio Compatibility Information](http://kyosho.com/mini-z-info/support/radio-compatibility-information/) which shows that the  KT-531P transmitter is using a different protocoll (FHSS) but is also based on the A7106 transmitter chip I tried to get the same information and logs as I didn't manage to bind to a multiprotocol module.
Maybe someone could take a look at it to tell if it is a new protocol, or a new subtype to something already known...
[KT-531P-24MHz.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/6325691/KT-531P-24MHz.zip)
[image]
[image]

Transmitter has throttle and steering, a potentiometer called CH4 which controlls the blink frequency if there are lights installed in the car and a potentiometer called ST.DR which affects steering. Trims for throttle and steering are also present but i would think these are offset to the channels.

pascallanger, 2021-04-16:
I'll have a look but I already have a folder with that TX name so most likely i looked at it already but failed to reverse it.
Let's see if a second TX would help to reverse it.

pascallanger, 2021-04-16:
This is the Kyosho/FHSS protocol. So I'm quite surprised that it doesn't work for you... Can you retry?
If it doesn't work, can you try with this line uncommented: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/756e9b3213a0b8cb514898cce926f86c4d0e6c96/Multiprotocol/Kyosho_a7105.ino#L20
Next if it still doesn't work, can you try this release when the protocol has been introduced: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/releases/tag/v1.3.1.49 , may be I broke it when adding the hype sub protocol or another change...

AndreasAusD, 2021-04-16:
I'm afraid you broke something :-)
I need to get the build environment running first, so instead I tested the old release 1.3.1.49
It is working, the car binds with steering on CH1, throttle on CH2 and the lights controllable on CH4
The steering travel adjust on CH3 ist not working for me, but I'm not shure if this is really send over the radio or not.

Updated again to 1.3.2.61, protocol KYOSHO - FHSS car won't bind

pascallanger, 2021-04-16:
Can you try the different releases to find in which one it breaks?

AndreasAusD, 2021-04-16:
yes, I can give it a try during the weekend.
That would only be the ones where Kyosho_a7105 was modified, right ?
So the releases before and the ones with the commit ?

pascallanger, 2021-04-17:
Yes please test each releases/push before and after each time the protocol has been touched.

Anyway, I've looked at it this morning and found one thing. Can you try v1.3.2.74 available here: https://downloads.multi-module.org/latest-test/

pascallanger, 2021-04-18:
Any test?

AndreasAusD, 2021-04-19:
sorry, haven't been able to test much as the ready build 1.3.1.65 was not working and I then realized there is a problem with the STM32 chip as I'm using a TMX5 4in1 module.
need to figure that out first...

AndreasAusD, 2021-04-20:
hi, a self compiled version 1.3.2.76 is working for me now

closed 2021-04-21 by pascallanger