pascallanger/DIY-Multiprotocol-TX-Module #564: Using a transceiver module directly from a computer

rmst, 2021-04-17:
Hey, I wanted to write you @pascallanger directly but couldn't find your email. Perhaps for the better, in case other people have the same question.

I'd like to control quadrotors with Frsky receivers from my computer. I understand that this repo is more geared towards being used with a remote control but would it be possible to also hook a transceiver like the CC2500 up to a computer via USB somehow?

kilrah, 2021-04-17:
No but you can connect the whole module if you use an usb-serial adapter and write software to talk to the module according to its protocol.
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Multiprotocol/Multiprotocol.h#L802

rmst, 2021-04-17:
Oh, so I could just buy a

- USB-to-serial adapter like https://www.aliexpress.com/item/1005001510943320.html
- CC2500 like https://www.aliexpress.com/item/32916163208.html

wire them up and the CC2500 is going to understand this protocol out of the box?

pascallanger, 2021-04-18:
That's not what he said... He said to use a multi module and connect it to a computer. Someone else as done this and there are details somewhere in a closed issue. I let you search for it.

closed 2021-04-18 by pascallanger