pascallanger/DIY-Multiprotocol-TX-Module #565: FrSky D16 should not automatically leave Bind mode

bitogre, 2021-04-18:
With some FrSky X-Series receivers, I am having problems getting the receiver to bind with my RadioMaster TX16s before it automatically leaves bind mode (button is just too small to reliably press and hold while plugging in power).  On my Taranis, it would remain in Bind mode until I told it to end binding.  It would be helpful if it did not automatically leave the bind mode or if there was a way to increase the time it would be in bind mode before it leaves.

pascallanger, 2021-04-18:
To bind with multi with whatever protocol, you must first put the RX in bind mode then launch the bind on the TX.

closed 2021-04-18 by pascallanger

bitogre, 2021-04-18:
That is not what the FrSky Receiver manual says.  You can see the procedure they state here in this example: https://www.frsky-rc.com/wp-content/uploads/Downloads/Manual/G-RX6/G-RX6%20ACCST%20-Manual.pdf

kilrah, 2021-04-18:
Manual can say whatever, the point is that it doesn't matter so there's nothing wrong with putting the recevier in bind mode first.