pascallanger/DIY-Multiprotocol-TX-Module #573: Binding DMS RX (Spektrum DX6i) does not work

drtrigon, 2021-04-25:
I have a TX16S with external multi-module (Jumper JP4IN1-SE) and try to use it for wireless trainer. I flashed the most recent version of the firmware (1.3.2.77) and opentx is 2.3.9

The multi-module does not bind to the DX6i. I selected DSM RX and tried with several receiver numbers. I also asked in the RCG forum and they essentially told me:
- select DSM RX
- choose any receiver number (not important)
- start binding on the master (DSM RX) side, so for me the TX16S
- start binding on the student, so for me the DX6i
- then I have 2 radios sitting there and beeping forever - no bind at all
- I tried several distances between the two radios from antennas touching to 1-2 meters - indoors and outdoors
- (I also selected Master/Multi as trainer mode)
- (I also tried several receiver numbers)

What confuses me is that the DSM RX page (https://www.multi-module.org/using-the-module/protocol-details/dsm-rx) clearly states "The same Receiver Number must be used on the trainer and trainee radios." and I have no idea what the number on my DX6i is and how I could figure that out?

What am I doing wrong?

pascallanger, 2021-04-25:
> What confuses me is that the DSM RX page (https://www.multi-module.org/using-the-module/protocol-details/dsm-rx) clearly states "The same Receiver Number must be used on the trainer and trainee [...]
The doc I maintain is here and there is nothing about the rx number just because it is fully ignored: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#DSM_RX---70

Usually people reporting issues to bind is due to the fact that the other module is still transmitting. Can you make sure that you are binding with only one module turned on, the other one should be off.

benlye, 2021-04-25:
Oops. I must have copied and pasted that into the doc site without checking. I'll fix it later today.

drtrigon, 2021-04-25:
> Usually people reporting issues to bind is due to the fact that the other module is still transmitting. Can you make sure that you are binding with only one module turned on, the other one should be off.

Thanks for your answer. Yes the doc here is quite different, good hint!

I tried with the internal module off right now and got the same result; no bind. What else?

pascallanger, 2021-04-25:
Have you tried with LNA on/off?

drtrigon, 2021-04-25:
Yes. I tried this some time ago, and right now again. No change...

pascallanger, 2021-04-25:
Next step would be to use debug firmware on the external module using flash multi with checking the box "run firmware after flash" and open the built in serial monitor, launch a bind on both and see what the module sees.

drtrigon, 2021-04-25:
Ok, I'll do so!

Let me get this right (flash ftdidebug or usbdebug, how to connect and run the test, etc.). Do you have a how to for this? I found https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Advanced_Debug.md but it looks a bit old to me. And I hesitate to put the powered-on module into the transmitter (or has the module to be in the transmitter already? do I need to turn the transmitter on or off?). And then I would have to run a serial console (arduino ide or other) anyways as I would flash the pre-built binary and not build it myself... etc. Furthermore I don't know what to do then in order to test properly? Just do the same as before; configure the module as external DSM RX and initiate binding? Whats the proper procedure to produce useful data for you?

pascallanger, 2021-04-25:
1. Download the FTDI debug fw
2. Launch flashmulti
3. Somewhere in the menu you have something called run firmware after flash, check it
4. Flash the firmware you downloaded in 1
5. In flash multi enable the serial monitor
6. Power on the radio
7. Select the dsm rx protocol on the external module, internal module off
8. Launch a bind
9. You should already see stuff in the serial monitor windows
10. Launch a bind on the dx6i
11. Wait and may be relaunch a few time the bind on the dx6i
12. Save the file of the serial monitor (export)
13. Upload the file here

I don't have my computer with me but that's the detailed steps. The debug firmware is a standard firmware with debug enabled, you won't break anything...

drtrigon, 2021-04-25:
Ok may be I should have mentioned that I am working on linux here.

Used a VB to run win and tested as you explain:
- I am not able to flash "multi-stm-xn297dump-ftdidebug-v1.3.2.77.bin" nor "multi-stm-serial-aetr-v1.3.2.77.bin" that way; whenever I start the TX afterwards I get "Invalid Protocol" error

Then I switched to linux (as I recognized an "executing" message there):
- flashing "multi-stm-xn297dump-ftdidebug-v1.3.2.77.bin" results in an "Invalid Protocol" error
- flashing "multi-stm-serial-aetr-v1.3.2.77.bin" works

pascallanger, 2021-04-25:
The dsm rx protocol is not most likely excluded from the automatic build since the debug firmware is much more larger and all protocols can't fit. Can you use the Arduino ide instead to compile your own debug version?

drtrigon, 2021-04-25:
Sigh. I'll have to see whether I can get this to work...

Another point; can yo confirm that actually the exactly same firmware file should work on the internal module too? I tried the firmware file from the internal module on this external, as the internal was able to connect through DSM RX. Strangely the protocols available are not listed in the same order (opentx issue) as on the internal DSM RX follows DSM but not on the external - even when the run the exactly same firmware.

Another point is I opened the module and found that inside there is a jumper labeled 1S and 2S but I could not find any reference on what the jumper is doing - actually all picture I found here do not show a jumper.

And a last point is that when I compare the lenght of the antenna to the internal one it is shorter. When I compare it to a 5.8GHz one from a video transmitter it has the same lenght...

I bought the module on aliexpress and do now get the impression something is not ok with the hardware...

Any thoughts?

drtrigon, 2021-04-26:
Finally! I figured it out (my last assumption turned out to be the right one); the first pin (PPM) is broken and does not connect anymore. So the issue can be closed as it is not about a firmware issue! Thanks for your support!

One additional question out of curiosity; how does the serial protocol work? Does it use 2 pins (RX, TX) or just 1 (RX/TX on same pin) - respective how are telemetry and other date sent from the module to the transmitter? Because this data seemed to come through. (If the protocol uses just 1 pin form both RX and TX can you give me some more info because I am intressted in that generally.)

closed 2021-04-26 by drtrigon

pascallanger, 2021-04-26:
The radio is outputing the commands on the 1st pin of the module bay and receives the telemetry+status on pin 5.

drtrigon, 2021-04-26:
Thanks for the info! 👍