pascallanger/DIY-Multiprotocol-TX-Module #574: LSR-3000 Protocol(may be old DSM Protocol for car) request

doohans, 2021-05-01:
Honorable pascallanger.
I sent this article to you by direct mail via rcgroups.(to midelic)

I really love the multiprotocol you made. And I will capture and send my old dsm protocol files. This protocol is used to control the car. It is not a DSMR protocol. Please add this protocol. The spi capture file is attached as a Google link.

[TX Info]
my tx is spectrum LSR-3000.
https://www.spektrumrc.com/Products/...rodID=LOSB0805

[SPI Capture Files]
1. bind
https://drive.google.com/file/d/1Z9M...ew?usp=sharing
https://drive.google.com/file/d/1NMd...ew?usp=sharing

2. forward
https://drive.google.com/file/d/1z-5...ew?usp=sharing

3. backward
https://drive.google.com/file/d/19iz...ew?usp=sharing

4.steer
https://drive.google.com/file/d/1zyn...ew?usp=sharing

Best regards,
doohans

pascallanger, 2021-05-01:
What's the RF component?
I don't have access to the files... I've made a request.

doohans, 2021-05-01:
i changed access files. please open them again.

pascallanger, 2021-05-01:
Something is not correct in the dumps. Please redo them at 12MHz or 16MHz instead of 8MHz which is too low.

pascallanger, 2021-05-01:
Also the dumps for throttle and steering are too small.
You need to move slowly and independently each input:
 - one file with steering mid to max right then max left then mid
 - one file with throttle mid to max forward then max backward then mid
Make sure there is no trim or dual rate set on the TX while doing this.

doohans, 2021-05-01:
ok. i re-dump spi capture files with 16Mhz option.

https://drive.google.com/drive/folders/16Kq-LmScoJGK6ErM9ImJxX5dkFxbmOmo?usp=sharing

i found something weird. 

i tried a couple of times. and i got same error on same position. 
[image]

and many errors occur during dumping. i think ENABLE signal is NOT stable.
[image]

I attached not only hex files but also .logicdata files. please check the files.

thank you.

doohans, 2021-05-02:
I didn't connect GROUND. 
Finally, I've got clean ENABLE signal with GROUND connection.

Please refer to the following files.

https://drive.google.com/drive/folders/1-u6e7i3Jdhp6qhw1Ur9FAB4QOkQaEmGh?usp=sharing

Thanks.

pascallanger, 2021-05-02:
It looks far better now. They use a mode with the cyrf that I haven't seen so far so my decoder is struggling, I'll see if I modify it to fully decode what's being sent.
They always transmit on the same channel. At power up they scan starting at 0x13 every odd channel until they find one which probably has the lowest low RSSI, 0x27 in the dump. Then they try 8 different data codes to most likely see if another TX is transmitting but this is an assumption.
After that they set one last data code and transmit bind packets every 8.8ms.
Bind data code:
0xD7 | 0xA1 | 0x54 | 0xB1 | 0x5E | 0x89 | 0xAE | 0x86 | 0xC9 | 0x2C | 0x06 | 0x93 | 0x86 | 0xB9 | 0x9E | 0xD7
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --

Bind packets:
0x47 | 0x52 | 0xAE | 0xAA | 0x47 | 0x52 | 0xAE | 0xAA | 0x05 | 0x52
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
ID[0]|ID[1]|ID[2]|ID[3]|ID[0]|ID[1]|ID[2]|ID[3]|??|??

In normal mode, they send 3 channels. The payload is somehow obsfucated. Most likely xored but not only since the bits seem to also not be in order.
The normal data code is:
0x5F | 0x30 | 0x3B | 0x56 | 0x96 | 0x45 | 0xF4 | 0xA1 | 0x03 | 0xBC | 0x6E | 0x8A | 0xEF | 0xBD | 0xFE | 0xF8
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --

And the data:
0x47 | 0x52 | 0xAE | 0xAA | 0x56 | 0x87 | 0x38 | 0x85 | 0x57 | 0xF6
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
ID[0]|ID[1]|ID[2]|ID[3]|ST|ST|TH|TH|CH3|CH3

pascallanger, 2021-05-02:
Can you do a dump where you move extra slowly the steering wheel?
Turn the steering full left, start to record and move extremely slowly until you reach full right.

doohans, 2021-05-02:
ok. starting with Full left steering.

https://drive.google.com/file/d/1CfjmJ8VtNx1vGHFijGw69UUfWpEEdLz8/view?usp=sharing https://drive.google.com/file/d/1YLHtLzsR0qRbKNwYSUnB7yXDOGxKJpCT/view?usp=sharing

doohans, 2021-05-02:
i can't remove trim value both throttle and steering. because there is only analog knob to adjust trim without any LCD screen on my TX. so i it's very hard to make it center value. i adjust them to center position in case of only for my LOSI TRAIL TRREKER.

pascallanger, 2021-05-02:
In the example above for steering, the value sent is 0x56 0x87.
0x687 is the steering wheel position which goes from 0x000 to 0xD1E, 0x68C being the middle which is not too far from the expected middle (0xD1E-0x000)/2=0x68F.
The problem is the high nibble 5 in the example above. It seems to be a sort of check of the attached value.
Example of the check where we see that the high nibble is xored by F when the bit3 of the value is changing:
0x57 | 0xF4
-- | --
0x57 | 0xF5
0xA7 | 0xF3
0xA7 | 0xF2

I'll now check the new dump to see if it becomes more clear.

pascallanger, 2021-05-02:
So far I've reversed the check for the first 8 bits. Still 4 to go...

pascallanger, 2021-05-02:
I have the 12 bits:
b0 | b1 | b2 | b3 | b4 | b5 | b6 | b7 | b8 | b9 | b10 | b11
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
0 | 0 | F | 1 | D | A | B | 6 | C | 8 | F | 1

If the bits are set then xor with the table value above and finish by a final xor of B.

pascallanger, 2021-05-02:
It could be written differently like b12=1^b2^b3^b4^b6^b10^b11, b13=1^b2^b5^b6^b7^b10, b14=..., b15=...
Not sure what's the most effective way to code it but I think I'll go with the table above.

doohans, 2021-05-02:
Wow you almost solved it. I am very excited and looking forward to it.

pascallanger, 2021-05-03:
Looking now at the cyrf config, they've disabled CRC. This is why they are adding these check/FEC like bits on the channel values.
Therefore I'm now wondering if the last 2 bytes of the bind packet could be a CRC. That would prevent us to change the ID if we have only your TX ID... So let's see.

pascallanger, 2021-05-03:
@doohans Please test v1.3.2.80 available here: https://downloads.multi-module.org/latest-test/
It uses your TX ID and settings for now so you don't have to bind the RX (but you can test it).
In the case it works 🤞, more tests will have to be conducted to check if we can change the ID and settings.

doohans, 2021-05-03:
GREAT!!. This works perfectly!!

pascallanger, 2021-05-03:
> GREAT!!. This works perfectly!!

Control works fine, correct?
Ok, have you tried to bind? Is the RX reacting correctly (blinking pattern)? With Multi you have to press the bind button on the UI (not at powerup like the original).
Are you able to compile the source code by yourself so I can guide you to do and test some modifications?

doohans, 2021-05-03:
It didn't work the first time I didn't bind.
However, after binding, it worked normally.
The binding appears to be working correctly.
I am using multi in PPM mode. So, I am using the source code. I can do simple coding.

pascallanger, 2021-05-03:
There are a few tests to be conducted.
First one is: in Losi_cyrf6936.ino, can you change this section: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/96b8a500b4f9c1e20f3b68dd0d09ab13ce431624/Multiprotocol/Losi_cyrf6936.ino#L127
From:
```
	#ifdef LOSI_FORCE_ID
		rx_tx_addr[0] = 0x47;
		rx_tx_addr[1] = 0x52;
		rx_tx_addr[2] =	0xAE;
		rx_tx_addr[3] =	0xAA;
		num_ch = 8;												// It looks that you could choose what you want based on channel business
	#endif

```
To:
```
	num_ch=(random(0xfefefefe)&0x07)+1;
	#ifdef LOSI_FORCE_ID
		rx_tx_addr[0] = 0x47;
		rx_tx_addr[1] = 0x52;
		rx_tx_addr[2] =	0xAE;
		rx_tx_addr[3] =	0xAA;
		//num_ch = 8;												// It looks that you could choose what you want based on channel business
	#endif
```

doohans, 2021-05-03:
The binding seems to work fine.
However, neither steering nor throttle works. My car doesn't move at all.

pascallanger, 2021-05-03:
Ok...
Try this:
```
	#ifdef LOSI_FORCE_ID
		//rx_tx_addr[0] = 0x47;
		//rx_tx_addr[1] = 0x52;
		//rx_tx_addr[2] =	0xAE;
		//rx_tx_addr[3] =	0xAA;
		num_ch = 8;												// It looks that you could choose what you want based on channel business
	#endif
```

doohans, 2021-05-03:
can't binding.

pascallanger, 2021-05-03:
Ok so without another radio (and the RX at hand), I have no idea how to add another ID...

doohans, 2021-05-03:
I have only ONE Losi TX.. we have to wait another person who can capture this TX signal...

Thank you very much for your hard work.

pascallanger, 2021-05-03:
Can you confirm the latest push to github is still working ok?

doohans, 2021-05-03:
It's not working correctly.

Binding succeeds, but no control.
If i  try it several times, it rarely gets in control.

pascallanger, 2021-05-03:
Hummm, I'm not sure this is due to the last push...
If you do get control after powercycling a few times, my guess is that the top rf channel is too high.
Can you try to change this line:
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/96b8a500b4f9c1e20f3b68dd0d09ab13ce431624/Multiprotocol/Losi_cyrf6936.ino#L124
to:
CYRF_FindBestChannels(hopping_frequency, 1, 0, 0x13, 0x27);

doohans, 2021-05-03:
same. binding is ok.
but i cant control both thottle and steer.
i can hear some servo noise when i turn steering wheel. but car doesnt move.

doohans, 2021-05-03:
i re-test with initial version you pushed. (i use latetest source replace only Losi_cyrf6936.ino file <96b8a500b4f9c1e20f3b68dd0d09ab13ce431624>)

same result. binding is ok. only few times i can control throttle and steering.
Even with success, if I turn the power off and on, I lose control again.

pascallanger, 2021-05-03:
Can you change CYRF_ConfigRFChannel(hopping_frequency[0] | 1); to CYRF_ConfigRFChannel(0x27);

doohans, 2021-05-03:
do i test with latetest version?  or initail version?

pascallanger, 2021-05-03:
They are basically the same in my eyes.

doohans, 2021-05-03:
it's working. binding is ok.
Even if the power is turned off, it works corretly. i can control both th and st.

~~~
void LOSI_init()
{
	LOSI_cyrf_init();
	CYRF_FindBestChannels(hopping_frequency, 1, 0, 0x13, 75);	// 75 is unknown since dump stops at 0x27, this routine resets the CRC Seed to 0
	CYRF_ConfigRFChannel(0x27);				// Only odd channels

	#ifdef LOSI_FORCE_ID
		rx_tx_addr[0] = 0x47;
		rx_tx_addr[1] = 0x52;
		rx_tx_addr[2] = 0xAE;
		rx_tx_addr[3] = 0xAA;
	#endif

	if(IS_BIND_IN_PROGRESS)
	{
		bind_counter = 300;
		CYRF_ConfigDataCode(LOSI_data_code[0], 16);				// Load bind data code
	}
	else
	{
		CYRF_ConfigDataCode(LOSI_data_code[1], 16);				// Load normal data code
		bind_counter = 0;
	}
}
~~~

doohans, 2021-05-03:
No binding with below code

~~~
	#ifdef LOSI_FORCE_ID
		//rx_tx_addr[0] = 0x47;
		//rx_tx_addr[1] = 0x52;
		//rx_tx_addr[2] = 0xAE;
		//rx_tx_addr[3] = 0xAA;
	#endif
~~~

pascallanger, 2021-05-03:
Leaving the module on and working, can you do a dump of power on of the original radio?

doohans, 2021-05-03:
is the module Multi module? so you mean that i turn on my multi module and original TX at same time?

pascallanger, 2021-05-03:
yes both on, first multi module then do a dump of the original TX boot

doohans, 2021-05-03:
ok i will test last success code.

doohans, 2021-05-03:
here dump.

[both_original_module.txt](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/6416090/both_original_module.txt)

pascallanger, 2021-05-03:
Can you do multiple dumps (at least 3) with multi on and the power on of the original TX?

Using the code from the first version, can you change this section: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/96b8a500b4f9c1e20f3b68dd0d09ab13ce431624/Multiprotocol/Losi_cyrf6936.ino#L127
From:
```
	#ifdef LOSI_FORCE_ID
		rx_tx_addr[0] = 0x47;
		rx_tx_addr[1] = 0x52;
		rx_tx_addr[2] =	0xAE;
		rx_tx_addr[3] =	0xAA;
		num_ch = 8;												// It looks that you could choose what you want based on channel business
	#endif

```
To:
```
	num_ch = (random(0xfefefefe)&0x07)+1;
	#ifdef LOSI_FORCE_ID
		rx_tx_addr[0] = 0x47;
		rx_tx_addr[1] = 0x52;
		rx_tx_addr[2] =	0xAE;
		rx_tx_addr[3] =	0xAA;
		//num_ch = 8;												// It looks that you could choose what you want based on channel business
		CYRF_ConfigRFChannel(0x27);				// Only odd channels
	#endif
```
Try multiple power on/off to see if you keep control all the time.

doohans, 2021-05-03:
I did the test you requested.
basically I use source code you linked.
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/96b8a500b4f9c1e20f3b68dd0d09ab13ce431624/Multiprotocol/Losi_cyrf6936.ino#L121

and i modified like below
~~~
void LOSI_init()
{
	LOSI_cyrf_init();
	CYRF_FindBestChannels(hopping_frequency, 1, 0, 0x13, 75);	// 75 is unknown since dump stops at 0x27, this routine resets the CRC Seed to 0
	CYRF_ConfigRFChannel(hopping_frequency[0] | 1);				// Only odd channels

	num_ch = (random(0xfefefefe)&0x07)+1;
	#ifdef LOSI_FORCE_ID
		rx_tx_addr[0] = 0x47;
		rx_tx_addr[1] = 0x52;
		rx_tx_addr[2] =	0xAE;
		rx_tx_addr[3] =	0xAA;
		//num_ch = 8;												// It looks that you could choose what you want based on channel business
		CYRF_ConfigRFChannel(0x27);
	#endif

	if(IS_BIND_IN_PROGRESS)
	{
		bind_counter = 300;
		CYRF_ConfigDataCode(LOSI_data_code[0], 16);				// Load bind data code
	}
	else
	{
		CYRF_ConfigDataCode(LOSI_data_code[num_ch], 16);		// Load normal data code
		bind_counter = 0;
	}
}
~~~

I did follow script.

1. I tried to bind with Multi Module until the binding was successful. I confirmed that I can control the car with the Multi. 
2. I kept the Multi powered on.
3. I Started to record dump.
4. then. I Turned On original TX. -> failed to control the car with original TX.
5. I tested it by repeatedly toggling the power of the original TX. 
6. I turned off power of original TX.
7. I stopped to record dump.

closed 2021-09-25 by pascallanger