pascallanger/DIY-Multiprotocol-TX-Module #576: 1.3.2.82 - Switching Banks in PPM Mode does not work

HazzaHFPV, 2021-05-06:
I have just flashed one of my iRangeX IRX4 Plus modules which I use on one of my old PPM radios to the latest master (1.3.282).
I just noticed that the switching between the 5 different banks does not seem to work anymore. Until now I only used the first bank, but I wanted to extend my use and define more protocols in the second bank.

I thought I might have done something wrong in my _config.h so I just did a clean configuration, and only changed a few things as the follwoing entries:

- #define TAER
- #define FORCE_FRSKYX_TUNING	-10
- #define FORCE_GLOBAL_ID	0x7146AE46
- removed several protocols that I do not need in the "PROTOCOLS TO INCLUDE" - Section
- //#define INVERT_TELEMETRY
- //#define INVERT_TELEMETRY_TX
- #define TX_CUSTOM
- #define MAX_PPM_CHANNELS 8 (8 Channel Radio)
- //#define TELEMETRY_FRSKYX_TO_FRSKYD

attached to the module is a bluetooth module (PIN5) for telemetry use. This does not seem to make any problems though, as I have confirmed the behaviour on three differen iRangeX modules.

I compiled via Arduino IDE and uploaded the resutling multi-stm-1.3.2.82.bin via flash-multi 0.6.1

When I select the last switch position on the module (yellow switch to "F") the green LED is solid and the red LED gives me a single flash, indicating Bank 1. When I press the bind key the red LED will remain on for about 1 second, which should indicate that it is switching to the next bank, but instead of blinking twice for bank 2, it remains in Bank 1 (still flashing only once).

I am not sure if it is an error that I made, or if I found an issue
I have attached my compiled multi-stm-1.3.2.82.bin and my _config.h
Maybe someone can spot my mistake or the issue

In the meantime I will download older PPM versions via https://downloads.multi-module.org/ to see if I can find the version where it stopped working, or if it is indeed a user error on my side.

Thank you for your help :)
[config.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/6437089/config.zip)

pascallanger, 2021-05-06:
I'm on my phone so I can't check your config file but the default number of banks is set to 1 and since I don't see it listed above i think this is your issue

pascallanger, 2021-05-06:
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/06b336ee597de3edb5bf28f1cc65858f70be8ae3/Multiprotocol/_Config.h#L423

HazzaHFPV, 2021-05-06:
Aaaahh.. what a silly oversight from my side. That was it!
I overlooked that line completely.
Just like I suspected: user error, but thank you for that quick hint :)
I just didn't spot it..

Thank you very much Pascal :)

HazzaHFPV, 2021-05-06:
@pascallanger 
maybe you could give me another quick hint. As the receiver number of a cloned radio has to be set to the same value as on the original radio, I am currently configuring my banks with the FrskyX and FrskyX2 cloned subprotocols, each position with a different receiver number (as I only have the yellow selection knob available to switch between different protocols or in my case different receiver numbers).

If I use the listed "XCLONE" Subprotocol for FRSKYX2 I fail to compile. Could you point me to the correct naming, or am I doing another silly mistake ? :)
[image]

Please see attached screenshot. I am not sure if the protocol-list in the config.h is quite correct as it also has some double-entries:

closed 2021-05-06 by HazzaHFPV