pascallanger/DIY-Multiprotocol-TX-Module #578: FRSkyX2 not binding to R161/R168

Enggmaug, 2021-05-13:
Hello, 

I do know these won't sell anymore, but I'd like to be able to use R161/R168 Receivers in X2 mode.

Radiomaster had released the firmwares to do so, for both FCC or LBT modes.

I first thought it would be solved in 1.3.2.77 with the evolution "FrSkyX2-Updated bind to work with some new RXs code".
But I still can't Bind.

I can Bind in FrskyX, when the appropriate firmware is loaded on the RX, but not in X2. 

Is there a way to make it work, please ?

Enggmaug, 2021-05-13:
Adding details, I use the internal module of my RS TX16S.

pascallanger, 2021-05-13:
Contact Radio Master. I don't have any of these RXs to test so I can't do anything. The X2 works fine with genuine Frskyx receivers so the issue is not on the multi side.

closed 2021-05-13 by pascallanger