pascallanger/DIY-Multiprotocol-TX-Module #579: R-XSR ACCST 2.1.1 only partial Telemetry

HazzaHFPV, 2021-05-14:
Not sure if this can be fixed on the multi-side, but there seems to be a difference in the ACCST 2.1.0 and 2.1.1 F.PORT firmware, at least on the R-XSR. With 2.1.1 only the receiver telemetry is found on the radio (Jumper T16), but none of the betaflight telemetry (current sensor, GPS, heading, etc.) is found or displayed.

On firmware ACCST FPORT version 2.1.0 (EU LBT) both the receivers own telemetry and Betaflight telemetry sensors are found and displayed.
Not sure if FrSky fucked up something again. Any thoughts?
Used on Betaflight 4.2.9 F405 (CLRACING F4S flight controller) with GPS and ESC telemetry.

MRC3742, 2021-05-15:
There seems to be a problem receiving telemetry using FPort on R-XSR  v2.1.1
Reverting back to version 2.1.0 is the only fix currently?
It's been reported numerous times to FrSky but no fix, acknowledgement or response yet.

https://github.com/FrSkyRC/Firmware-Test/issues/118

Rich

HazzaHFPV, 2021-05-16:
Ok, thank you for your reply @MRC3742
..so it seems to be another FrSky problem apparently and not an issue with the MPM-Firmware.

closed 2021-05-16 by HazzaHFPV