pascallanger/DIY-Multiprotocol-TX-Module #583: Flashing 1.3.2.77 TAER to Jumper T-Lite gives "No Telemetry" - no protocols.  Flashing back to AETR order works.

Zebble, 2021-05-23:
Running Open-TX 2.3.12N504 (and was on N501) and everythin works with AETR order firmware for the multi-module.  As soon as I change it to TAER I get "No Telemetry".  I've also made sure the model/radio is using TAER but it shouldn't matter until the Multi-module is recognized, correct?

I've used a similar setup with another radio (RadioMaster and Horus X12S with a multi-module) without issue.

benlye, 2021-05-23:
Exactly which firmware file did you flash?  I just put multi-stm-serial-taer-v1.3.2.77.bin on my T-Lite and it worked fine.

You didn't accidentally flash a PPM-mode instead of the serial file?

Joetrin, 2021-05-23:
That's the exact same file I used.  Flash it and it goes to "No telemetry".  Flash it back to multi-stm-serial-aetr-v1.3.2.77.bin and it works as expected.

benlye, 2021-05-24:
Corrupted downloads?

The MD5 hash of the file should be `B16C116E9561883BC7A477BD04F078E2`

Zebble, 2021-05-24:
That was the problem.  The download itself wasn't corrupt, but copying it to the radio and it became corrupt.  Sounds like a flaky SD card.  I copied it again as a new file instead of over the old file, flashed it and it worked fine.

I had downloaded it from multi-module.org several times thinking it might have been a corruption issue.  I didn't think to do the same tests on the SD card itself!

Thanks for the help.  Will close.

closed 2021-05-24 by Zebble