pascallanger/DIY-Multiprotocol-TX-Module #586: Irx4 lite no serial input

rayb106, 2021-06-03:
Hi All I’m new to droning and just bought a tiny hawk s I also have a 5”Quad which I can connect to easily, but I’m trying to bind to my tiny hawk with the irx4 lite on my frsky xlite pro, I’ve noticed after updating the firmware I’ve lost the multi option and the irx4 has a solid green light and slow flashing red light, it seams to have no serial input now, I’ve also updated the frsky firmware too for the frsky site

pascallanger, 2021-06-03:
The radio must be updated with OpenTX not a firmware from frsky which doesn't support the multi module. You also need to make sure that the OpenTX build you use is including multi sport when selecting it in companion TX.

rayb106, 2021-06-03:
> The radio must be updated with OpenTX not a firmware from frsky which doesn't support the multi module. You also need to make sure that the OpenTX build you use is including multi sport when selecting it in companion TX.

Thanks for the reply, how do I know if it includes multi sport when selecting ? Also I can't remember how to get the boot loader up from start on the frsky pro, I've held all the buttons on the d-pad while turning on

pascallanger, 2021-06-03:
Download companion TX, when selecting your radio you're there will be some choices with one of the choices being multi then power of your radio, plug it to the computer and flash it within companion TX. There are plenty of videos on the web to do this.

closed 2021-06-03 by pascallanger

kilrah, 2021-06-03:
There's no option anymore, multi support is always in.

rayb106, 2021-06-03:
I noticed this when trying to set up a new radio, just got to find the right firmware now for my frsky xlite pro

rayb106, 2021-06-04:
Thanks guys all sorted out now