pascallanger/DIY-Multiprotocol-TX-Module #587: Support for RBRC Receiver

drtrigon, 2021-06-04:
I have two multiprotocol modules and wonder whether it is possible to support RBRC receiver, like: https://www.banggood.com/RBRC-Receiver-Circuit-Board-for-RB1277A-1-or-12-RC-Vehicels-Model-Spare-Parts--p-1596090.html?cur_warehouse=CN&rmmds=search

This board is the receiver (including esc) for: https://www.banggood.com/RBRC-RB1277A-1-or-12-Drift-RC-Car-with-Mecanum-Wheels-2_4G-4WD-4CH-High-Speed-35km-or-h-Electric-Full-Proportional-RC-Model-Toy-for-Beginner-and-Kids-p-1545811.html?cur_warehouse=CN&ID=6157196&rmmds=search

I figure it is hard to judge just from the info available. I ordered one of those cars ;) and want to ask what is the recommended (simple) way of figuring out whether one of the already existing protocols works together with this board? (Given the number of supported protocols is in the order of 100, I would like to avoid having to test every one of them.)

pascallanger, 2021-06-04:
I think that I can see the RF chip marking which looks to be a xn297l so you can narrow your research to only the xn297 protocols and associated sub protocols.
If you can't find one working and if you have an external module then we can try to reverse the protocol.

drtrigon, 2021-06-04:
Sound great thanks!

Where can I find a list of the xn297 protocols and associated sub protocols?

pascallanger, 2021-06-04:
Last column of the protocol table: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#available-protocol-table-of-contents-listed-alphabetically

drtrigon, 2021-06-04:
Thanks! (and sorry - I was searching through the page but had a space at the end of the string so it didn't trigger...)

closed 2021-09-25 by pascallanger