pascallanger/DIY-Multiprotocol-TX-Module #595: Can't flash my board, no serial pins (tx/rx) and gdt

Spolverino, 2021-06-28:


pascallanger, 2021-06-28:
?

Spolverino, 2021-06-28:
sorry this is my first time commenting on github

pascallanger, 2021-06-28:
Still, no idea what is your problem... Which module? which current version? which radio? which fw on the radio?...

Spolverino, 2021-06-28:
My module is the jp4in1. I don’t know what the firmware version is but there’s a v1.2 on the pcb.
[image]

Spolverino, 2021-06-28:
I am using an fs i6x’s ppm output and feeding it to the module

pascallanger, 2021-06-28:
To update the module follow: https://www.multi-module.org/using-the-module/firmware-updates#update-process

closed 2021-06-28 by pascallanger

Spolverino, 2021-06-28:
already tried all the methods avalable to me (all except the radio one since mine doesn't support it)
[image]

Spolverino, 2021-06-28:
And since my board doesn't have the rx and tx pins I can't update my bootloader as already said.

pascallanger, 2021-06-28:
Enable verbose output.
My guess (from the little the arduino window is telling) is that your antivirus as quarantined one file which is needed to flash...

pascallanger, 2021-06-28:
> And since my board doesn't have the rx and tx pins I can't update my bootloader as already said.

None sense comment on this module...

Spolverino, 2021-06-28:
Looking for Maple device ...

No MULTI-Module device found.

Probing serial port COM12 for STM32 in BOOT0 mode ...
Failed to init device.
stm32flash 0.4

http://stm32flash.googlecode.com/

Interface serial_w32: 115200 8E1

Module in BOOT0 mode not found on COM12

pascallanger, 2021-06-28:
That's not the verbose output of FlashMulti.
Also above screenshots show COM13 and this one is COM12...

Spolverino, 2021-06-28:
yes it's the verbose output from the arduino as you requested. 
Enabled it on the FlashMulti now.
For the COM port: I tried using the usb connection to my module to program it. Had used different arduinos before and the result was still the same.

.\tools\stm32flash.exe -b 115200 -r "C:\Users\nicol\AppData\Local\Temp\tmp5891.tmp" COM13

stm32flash 0.4
Failed to init device.

http://stm32flash.googlecode.com/

Interface serial_w32: 115200 8E1

benlye, 2021-06-28:
Try different USB cables, different USB ports, and if you can, a different computer.

Spolverino, 2021-06-28:
Have already tried all of that sadly :/ Even went to a friend to test it on his computer

Spolverino, 2021-06-28:
is the cp2102 ic in charge of USB programming? Could it be possible to manually solder some wires to it's rx tx pins in order to flash the bootloader?

benlye, 2021-06-28:
Just looked closely at the image - you have 3V3 bridged to BOOT0.  You should remove that so that you can use the USB port.

benlye, 2021-06-28:
> is the cp2102 ic in charge of USB programming? Would it be possible to manually solder some wires to it's rx tx pins in order to flash the bootloader?

If you want to try something like that you could try to solder directly to the RX and TX pins on the STM32F1, bypassing the CP2102 entirely.

Spolverino, 2021-06-28:
Yes I removed it while programming from usb

> Just looked closely at the image - you have 3V3 bridged to BOOT0 with solder. You should remove that so that you can use the USB port.

Spolverino, 2021-06-28:
> > is the cp2102 ic in charge of USB programming? Would it be possible to manually solder some wires to it's rx tx pins in order to flash the bootloader?
> 
> If you want to try something like that you could try to solder directly to the RX and TX pins on the STM32F1, bypassing the CP2102 entirely.

Yes I'll try this way. I have an stlink to flash the stm

benlye, 2021-06-28:
> Yes I'll try this way. I have an stlink to flash the stm

None of our programming tools (neither Arduino IDE or Flash Multi) support using an ST-Link, so you're on your own with that.

Spolverino, 2021-06-28:
Is there an alernative which works with Arduino or flashmulti?

benlye, 2021-06-28:
> Is there an alernative which works with Arduino or flashmulti?

Yes, an external USB-to-serial adapter connected directly to the RX and TX pins on the STM32.  I would usually recommend anything with a CP2102 on it.  Just make sure whatever you use operates at 3V3.