pascallanger/DIY-Multiprotocol-TX-Module #599: R-xsr receiver

rayb106, 2021-07-16:
Hi All,

I'm trying the bind my new r-xsr mini d16 receiver with my frsky (Eu FW from frsky site)lite pro x ( has the latest open tx EU) rx1 on mamba f722 mini set up as rx port (latest fw from mamba site). 

I'm following the bind instructions,when connected to power the receiver starts up with solid blue light and red flashing light, so it's in Sbus mode and not binded with anything. When holding the f/a button and connection power through a smoke stopper it then has  the green light is solid as well as the blue and red, radio is in bind after the quad is powered and f/s is held in, once I take my radio out of bind and power down the quad, when I re connect it the red is still flashing and blue light is solid. 

Anybody else has this issue, need to get the Atto off the ground 😭😭

HazzaHFPV, 2021-07-18:
What file did you flash on the R-XSR exactly?

rayb106, 2021-07-18:
Hi Harry,

I've sort it now I had to flash the r-xsr with the older firmware, as I've seen the new r-xsr come with new firmware which stops it working the my frsky-lite x pro, but all working now thanks

HazzaHFPV, 2021-07-18:
Awesome 👍

closed 2021-07-18 by pascallanger