pascallanger/DIY-Multiprotocol-TX-Module #600: Runcam nano 2 racer and runcam tx200u

rayb106, 2021-07-18:
Hi All,

I've got the above and have it working but I'm not seeing any think in the osd I am getting video feed and can see through the runcam on my goggles but I'm not seeing the name of craft, battery power,video channel, I have a matek F722 mini se, the tx200u vtx is connected to t4, 5v and G (to the left of T4) then the nano 2 connects into the vtx but I still have a port on the back which says menu which I've wired to the vtx port, but still get no info through my goggles. 

Please see pics attached   [image] [image]

pascallanger, 2021-07-18:
I don't want to be rude but what is the relation with multi on this issue?

closed 2021-07-18 by pascallanger

rayb106, 2021-07-18:
Ah sorry I though I was creating a new issue

HazzaHFPV, 2021-07-19:
Yepp you created a new issue, but it has, unfortunately, nothing to do with the multi-protocol Module. I might be able to help though if you drop me a short email (email can be seen in my profile)