pascallanger/DIY-Multiprotocol-TX-Module #601: How to flash DSMX only ?

andika207, 2021-07-30:
I want to know if it's possible to flash individual protocols (DSMX only) on the Lite type MPM modules ?
As you all know the ETHOS doesn't support MPM modules and there is no Lite MPM module on the market with analog dial for selecting protocols.

pascallanger, 2021-07-30:
You need to compile the code by yourself:
 - set all the ppm parameters in _config.h
 - set the first protocol of bank 1 to dsmx in the table in _config.h
 - uncomment this line to select dial 1: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/1a921b1cdb9947707475211d53c324ee60cff75c/Multiprotocol/Multiprotocol.ino#L513

The problem will be to bind unless your module has a button... A work around would be to load a firmware with autobind on when you need to bind and revert  when not binding.

andika207, 2021-07-30:
OK. I won't compile anything because I have no idea and none of the Lite modules come with a bind button that I know of. (Vantac, Jumper, Uruav, IRangeX, TBS) 

so the only real alternatives are the Lemon DIY module or a custom made mod with a bind button.
- https://lemon-rx.com/index.php?route=product/category&path=84
- https://www.rcgroups.com/forums/showpost.php?p=47543499&postcount=9895

pascallanger, 2021-07-30:
Some lite modules do have a bind button at least the TBS does. And adding it to any module is not that difficult...
Compiling is extremely easy.
But if you don't want to do anything by yourself and DSM is the only thing you want then the Lemon module is for you.
Pascal

closed 2021-07-30 by pascallanger

andika207, 2021-07-30:
> Compiling is extremely easy.

extremely easy for you... if you make a custom firmware for the TBS MPM module I would like to try it out.

pascallanger, 2021-07-30:
Have you even tried to follow the documentation? https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Compiling_STM32.md