pascallanger/DIY-Multiprotocol-TX-Module #602: Channel order issue w/ TX16s & Crossover XR602T-F2 rx

Daniel366Cobra, 2021-08-01:
Transitioning from Radiolink to OpenTX. The Crossover receiver is using FrSky D16 protocol. Bound without problems.

Using latest stable FW for OpenTX (2.3.14) and MM (1.3.2.86). Configured my TX to have inputs, mixers and outputs all in AETR order, the MultiModule FW is also AETR. However, the outputs come from the RX (as they have to be set in Betaflight) in ETAR order.
When reordering mixers to TAER while keeping the MM firmware at AETR, the correct order in Betaflight becomes AETR.

I tried searching the forums but haven't found a comprehensive answer,
I feel like there is a simple solution I overlooked somewhere. Please help.

pascallanger, 2021-08-09:
Which protocol are you using? Please post a pictures of the protocol page and mixers.
If the protocol is doing a transformation between input/output you have an option to disable through the GUI using the Disable Channel map box.

Daniel366Cobra, 2021-08-09:
I use "Frsky X D16" under internal RF. I believe it does not have the checkbox for disabling channel remapping, at least I have not seen it.
Here are the photos of my channel order and MM protocol.
 
[image]
[image]
[image]
[image]

HazzaHFPV, 2021-08-09:
Have you also selected the correct "Default channel order" within the OpenTX  Radio Setup?
[image]

Daniel366Cobra, 2021-08-09:
> Have you also selected the correct "Default channel order" within the OpenTX Radio Setup?
> Because on mine it works just as it should when everything is selected correctly (Frsky D16 as well as all other protocols)
> [image]

Yes, mine is also set to AETR there.

closed 2021-09-25 by pascallanger

timtuxworth, 2021-09-29:
Don't mess with the channel order on a multi-protocol radio. The module expects the channel order (usually AETR) and you will break things if you change it. I'm not sure about FrSky but for Spectrum, the module actually translates all your AETR inputs and mixes to TAER before sending to the receiver. If you change the transmitter you mess that up.
https://youtu.be/QUcCOqLcOj0