pascallanger/DIY-Multiprotocol-TX-Module #603: Switching models that use MPM in Tango2 requires power cycling radio

pashky, 2021-08-03:
Hi,

I've got Tango 2 on FreedomTX 1.29 and iRangeX IRX4 Lite module on 1.3.2.86. I have a model created for each protocol that I use (FrSky D16, Redpine, internal CRSF). So whenever I switch the model from MPM to another MPM I have to turn radio off and on again, otherwise status turns to "No input" and no transmission happens. Switching to internal crossfire and back to any external model works too. 

Now I understand there's too many variables and strongly suspect it could be FreedomTX or even upstream OpenTX issue rather than MPM, but I have no other hardware to verify that. So I'm wondering if anything like this have been reported before?

pascallanger, 2021-08-03:
Hi,
As far as I'm aware this is specific to FreedomTX. OpenTX/ErskyTX/er9x are working fine changing protocols on the fly.
Pascal

pashky, 2021-08-03:
Ok, I see. Thank you Pascal for confirming. Well, I guess I can only hope that mainline opentx tango2 support is finally coming soon.

closed 2021-08-03 by pascallanger